Misiu/argon40 #1: Enable the addon breaks HA API

welcoMattic, 2020-06-22:
## Version of the custom_component
0.0.1

## Configuration

```yaml
#...
argon40:
#...
```

## Describe the bug
I install the addon from HACS, as a custom repository.
I check that the addon is installed in custom_components directory (it is)
I add `argon40:` to my configuration.yaml file
I restart HA
HA won't start, the UI is still unavailable

## Debug log
Log level set to:

```yaml
logger:
  default: critical
  logs:
    homeassistant.core: error
    homeassistant.components.api: debug
    custom_components.argon40: debug
```

`docker logs homeassistant`
output:
```text
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[14:08:15] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2020-06-22 14:08:23 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-06-22 14:08:23 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for argon40 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-06-22 14:08:31 INFO (MainThread) [custom_components.argon40]
-------------------------------------------------------------------
Argon40
Version: 0.0.1
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/Misiu/argon40/issues
-------------------------------------------------------------------
```

If you need more logs, or logs from another location, tell me.

Misiu, 2020-06-22:
@welcoMattic thanks for the report.
I've just [pushed a quick change ](https://github.com/Misiu/argon40/commit/ef05d693526fded6428589e61b5b4dcca47368e1)that eliminates unneeded (for now) GPIO code.
I can't check that right now, but I'll try to test everything on my test env in the evening. Sadly I'll be able to check if Home Assistant works when I add this integration, but I can't check if the fan will work - I'm still without a case.

welcoMattic, 2020-06-23:
It works! I just installed the update via HACS. The fan works!

welcoMattic, 2020-06-23:
I just tried the automation, it works too! Thanks for the update.

closed 2020-06-23 by welcoMattic

Misiu, 2020-06-23:
@welcoMattic awesome news!
I'll try to finish work on the integration (hard to do without a case, but it's working 😁 ).
I'll notify you when I'll create a release.

welcoMattic, 2020-06-23:
I can help for testing before release if you need, feel free to ping me ;)

Stewie3112, 2020-07-11:
Same here, 
`The following integrations and platforms could not be set up:

argon40
Please check your config.`

the code pulled via git is different from the "quick change" you posted some days ago

Misiu, 2020-07-11:
@Stewie3112 I just checked the version that is installed via HACS and if you enable I2c on your PI it works correctly.
I'm working on an improved version that will support config flow, but by one year old son is taking all of my free time, so please give me some time to test everything.

syedkami, 2020-07-12:
Thanks @Misiu  for your gr8 effort. its working fine for me.
your component added through HACS.
otherwise created folder in config/custom_components/argon40. and copied all four files.
i was struggling hard to enable i2c.  But i did that directly added from here
folder in HA /etc/modules-load.d
file added in above folder:  rpi-i2c.conf
entries in above file:
i2c-dev
i2c-bcm2708

after, pulled out sd card and on windows laptop, edit config.txt file and uncomment below txt 
dtparam=i2c1=on
dtparam=i2c_arm=on

and now all working gr8. no error for argon40 on startup.
when you get succesfully all then you can find in
Developer tool/ Service Tab then argon40 fan speed in Service.

Thanks alot. now waitng for Power button ;-)

Stewie3112, 2020-07-12:
> @Stewie3112 I just checked the version that is installed via HACS and if you enable I2c on your PI it works correctly.
> I'm working on an improved version that will support config flow, but by one year old son is taking all of my free time, so please give me some time to test everything.

I enabled i2c following this [LINK](https://community.home-assistant.io/t/i2c-not-working-on-the-pi4/160489/3) , I'll try also syedkami's method this afternoon. Anyway thanks a lot for your great work, my son is 7 months and I can understand your lack of time XD

Misiu, 2020-07-12:
@Stewie3112 the method you described works on windows? I see only one partition when I plug the card.
I've tried enabling I2C via USB thumb drive but I had success only once.

I'd like to have a consistent method because I want to add it to the readme, so everyone won't have problems like we did.

Misiu, 2020-07-12:
@syedkami button is already supported (by home assistant) :)
Take a look at https://community.home-assistant.io/t/error-on-installing-a-script-for-case-fan-control-and-power-button-from-command-line/143182/39?u=misiu

Stewie3112, 2020-07-12:
@Misiu I used a virtualized ubuntu desktop to mount the other partitions

Stewie3112, 2020-07-12:
still no luck even with @syedkami method.
log says:

`2020-07-12 10:39:20 ERROR (MainThread) [homeassistant.setup] Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'`

then I guess it's definitely a problem with i2c not enabled

syedkami, 2020-07-12:
@Stewie3112  i had same error "No such file or directory: '/dev/i2c-1'",
but after doing these steps, all well.
first go to file editor in HA, then /etc/modules-load.d
[image]

create a file rpi-i2c.conf
and write in this file below txt
i2c-dev
i2c-bcm2708

then pulled out sd card.
explore in wondows
you will find config.txt file
open it with any file editor like notepad++
and uncomment bellow txt and save.
#dtparam=i2c1=on
#dtparam=i2c_arm=on

hope will solve issue.

Stewie3112, 2020-07-12:
I honestly don't know why, but at the third time editing the file via ssh in /etc/modules-load.d it's finally working!

the file gets erased (or moved?) at reboot the system, but once enabled seems to persist even if the file isn't there

Misiu, 2020-07-12:
@syedkami I didn't think about using `File editor` for that.
Wil test that method right away!

Stewie3112, 2020-07-12:
I used the cli to create and edit the file, but the result is the same. I can't use anyway the file editor to edit the root, or other than the /config folder. Is there a particular configuration to the plugin?

Misiu, 2020-07-12:
@Stewie3112 are the changes permanent? This won't roll back after a restart?
I had a problem with the USB thumb method - the config was imported (I was able to use my integration), but after a restart, the I2C was again disabled.

syedkami, 2020-07-12:
Yeah from  /etc/modules-load.d folder this file removed 
rpi-i2c.conf. but still working with no error.
[image]

Stewie3112, 2020-07-12:
> @Stewie3112 are the changes permanent? This won't roll back after a restart?
> I had a problem with the USB thumb method - the config was imported (I was able to use my integration), but after a restart, the I2C was again disabled.

I can confirm, once enabled. they persist a reboot of the host

Misiu, 2020-07-12:
@syedkami @Stewie3112 thanks for checking and for confirming!

syedkami, 2020-07-13:
Hi @Misiu ,
I am facing issue with this casing. 
Not regarding component. Thats working gr8.
When power cutts of, or loadshedding or by other means electricity gone.
Then have to press power button from back side manually 
If i am outside and it happens then will have to come back home and press power button. 
Have any solution.
Casing quality is awesome and your component is gr8, but this issue is annoying me alot.

Misiu, 2020-07-13:
@syedkami sadly no.
I'm going to contact Argon ONE and ask them for any hint on that. Ideally, there should be a switch inside that would allow choosing power mode - power by button or instant start.

syedkami, 2020-07-13:
I think there should be option that if we properly shutdown device then will start again unless press power button manauly. But if sudden power down then should start without it. 
I will also contact them today about this issue, because its a serious issue that you not at home and power down, then unable to do any anything.

Misiu, 2020-07-13:
@syedkami let me know if you get any response.

Stewie3112, 2020-07-13:
> I think there should be option that if we properly shutdown device then will start again unless press power button manauly. But if sudden power down then should start without it.
> I will also contact [...]

I hadn't considered this eventuality, looking too for a response if you get any!

welcoMattic, 2020-07-13:
Could you please open a new issue per subject? This one is done and closed.

syedkami, 2020-07-14:
Response on chat @Misiu 

[image]

digitalMedic, 2020-07-21:
I still cannot get I2C enabled.  I followed the instructions above and created the file rpi-i2c.conf in the /etc/modules-load.d folder and edited config.txt, but it still says it cannot set up Argon40.  When I went to edit the config.txt, I did not find a line that said #dtparam=i2c1=on.  So, I added it without the #.  I still get the following in the logs.  Any ideas?

`Logger: homeassistant.setup
Source: custom_components/argon40/__init__.py:64
First occurred: 11:41:33 PM (1 occurrences)
Last logged: 11:41:33 PM

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 34, in async_setup
    bus = SMBus(1)
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 193, in _async_setup_component
    result = await asyncio.wait_for(task, SLOW_SETUP_MAX_WAIT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/config/custom_components/argon40/__init__.py", line 64, in async_setup
    address,
UnboundLocalError: local variable 'address' referenced before assignment`

welcoMattic, 2020-07-21:
@digitalMedic Please open a new issue. This one is not related to your problem and it is closed.