Misiu/argon40 #2: Power  Button Issue

syedkami, 2020-07-14:
**Is your feature request related to a problem? Please describe.**

If Electricity sudden cuts off, then it wont start automatically.

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

i have purchased argon 40 casing for raspberry pi 4B, 
fully satisfied.

i have just one issue, if i am outside home and power or electricity goes, then i am unable to access my server untill and unless i come home and press power button manually.
is there any way that if i properly shutdown then manually press power button is ok, but if sudden electricity suppy disconnect then start resume last state, mean boot server without pressing any button.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Any way to disable power button.
Because i have chat with them, they clearly said its protection feature. In nexr version they will include this feature to start automatically.
[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

syedkami, 2020-07-14:
I have chat with argon they said you can control through i2c code. And send below link

https://github.com/Argon40Tech/Argon-ONE-i2c-Codes

syedkami, 2020-07-14:
4. HEX CODE: 0xfe

Set the ARGON ONE to Mode 2：

ACTION: Sets "Always ON Mode"

Behaviour: Power current will flow directly to Raspberry Pi. NO need to PRESS button to power ON from shutdown or power outage.

Misiu, 2020-07-14:
@syedkami this is interesting! I'll try to add this to my integration.

Did you see the issue in the repo you linked: https://github.com/Argon40Tech/Argon-ONE-i2c-Codes/issues/1?
Not sure if this will be possible, but give me some time and I'll try this out

Misiu, 2020-07-14:
@syedkami according to the issue some (older) cases don't support those codes.
Do you know how to tell the difference (visually by looking t case)? Also, do you know if users can update the firmware in the MCU?

Stewie3112, 2020-07-14:
following

Stewie3112, 2020-07-16:
Stupid question, but couldn't you try to add the script a service to enable or disable the power button as stated in the issue?

` 0x01a 0xfe`

I tried a `bus.write_byte(0x01a, 0xfe)` from HA ssh but I just get an "unexpected token" error

Misiu, 2020-07-17:
@Stewie3112 sure thing. Give me some time, I'll try to do that during the weekend.
I'm working on an improved version that will support config flow, so if the service works I'll add this as an option to the device

Misiu, 2020-07-21:
@Stewie3112 I just pushed changes to master branch.
You can update my integration using HACS.
Give me an update if that works with your case.

syedkami, 2020-07-21:
Hi @Misiu . 
I just reinstalled master branch from Hacs.
After restart, i pluged off my rp4. Then again on power back, had to press power button manually to start rp4.

Misiu, 2020-07-21:
@syedkami did you use the new service?
there should be `argon40.set_mode` service available.
You must call it only once. Sadly there is no way to tell if your case supports that command.

syedkami, 2020-07-21:
Yeah. I called this service and alway on = true.
Then unplug rp4. But still is same.
Now lets try other user then you will have idea what to fix.

Misiu, 2020-07-21:
@syedkami as wrote before not every Argon ONE case supports these commands. Chances are you have an older case and commands won't work.
I'll check the command on a second case when it arrives to check if my service works.
you can: buy a new case that supports these commands (there is no visual difference) or wait until Argon releases a new case that will have a jumper to select the mode (that will solve the problem once and for all)

Stewie3112, 2020-07-21:
Thanks a lot, I'll try this evening and report back the results

Misiu, 2020-07-21:
@syedkami more details here: https://github.com/Argon40Tech/Argon-ONE-i2c-Codes/issues/1#issuecomment-652142044
@Stewie3112 looking forward to an update from you

syedkami, 2020-07-21:
Thanks @Misiu .. i really appreciate your work.
Yeah may be. But i got my casing last week.
Anyways, will search to upgrade firmware. If you have idea how to update then share it and document link.
Thanks a lot.

Misiu, 2020-07-21:
@syedkami sadly the firmware update isn't possible. That's the info I got. But you can always ask Argon40 support.
Maybe you will have more luck than I did.
I'll be grateful for an update on that and more than happy to add a description to the repo.

syedkami, 2020-07-21:
@Misiu . I noticed onething that my fan is working fine at set speed. But when i unpluged and give power, then fan stop untill boot home assistant fully. Then start fan at set speed.
May be whatever component we write, it works after boot HA.
In my view that may be this command will add  in config like we did to enable i2c.

Misiu, 2020-07-21:
@syedkami my component requires fully running HA to be able to send commands to the case (to set fan speed).
However, the two commands `0xfd` and `0xfe` are special and need to be sent only once (the MCU remembers them). At least that's what I think :) I'll confirm that with Argon 40 support.

Misiu, 2020-07-21:
@syedkami Argon40 support confirmed that this command is sent only once.
All cases bought in March 2020 onwards should support that command.

syedkami, 2020-07-21:
Ok. Thanks. 
Now let see other user to experience. 
Is that same with others or just with me.

Stewie3112, 2020-07-21:
aaaaaand.... it's working!!! @Misiu just tried twice to be sure and now the pi boots once the power is back without having to press the power button. I think I've been lucky (case bought 2 weeks ago). Once again thanks a lot for your great work

syedkami, 2020-07-21:
@Misiu @Stewie3112  me still struggling :-(
Anyways gr8 effort @Misiu  really appreciated

Stewie3112, 2020-07-21:
@syedkami I'm sry for your bad luck, are you sure you have sent the command right? I attached a screenshot of what worked for me. Sent only once with no reply and nothing showed in logs

[image]

syedkami, 2020-07-21:
@Stewie3112  yeah, did that same. 
May be bad luck. Anyways will see later solution

Misiu, 2020-07-21:
@Stewie3112 thanks for checking :) Happy it works!
@syedkami it looks like you need to change your case. New version of it, that will have a physical jumper to choose the mode, will be released soon.
You can always sell your old case or use it for other projects.

BTW guys could you check the double-tap event? https://github.com/Misiu/argon40#bonus---button-double-tap-detection

Stewie3112, 2020-07-21:
> @Stewie3112 thanks for checking :) Happy it works!
> @syedkami it looks like you need to change your case. New version of it, that will have a physical jumper to choose the mode, will be released [...]

sure, copying your example automation works fine!

[image]

Misiu, 2020-07-21:
@Stewie3112 thanks! Now I must figure out why double tap works and single click doesn't.
Next step is config flow. When it's ready I'll create a new issue and ask for help with testing.

Torrecilla, 2020-09-28:
Hi, I'm facing an issue with the power button. I'll explain my setup:
I got my Argon 40 installed on a pi3B+
HA updated to 0.115.3 and everything else is working, HACS installed and some other addons working.
At first the Argon40 addon wasn't working but after trying all i2c enabling methods I think I got it working, is there any way to check if i2c has been successfully enabled?

Now: the fan is working and looks like I can control it from dev with the service "argon40.set_fan_speed" didn't tried the automation yet

Issue: The service "argon40.set_mode" doesn't appear on the list of services and tried your automation about double-tap but got no answer from HA

Misiu, 2020-09-28:
@Torrecilla I don't know if it is possible to check if I2C is enabled, but if you can control the fan from dev tools, then the dan is working fine!
Install my addon using the main branch, not the 0.0.1 release.
I'm waiting for a PR in HA Core repo that will allow me to rewrite this integration and add visual controls (control will be exposed as a fan UI)

Torrecilla, 2020-10-04:
> @Torrecilla I don't know if it is possible to check if I2C is enabled, but if you can control the fan from dev tools, then the dan is working fine!
> Install my addon using the main branch, not the [...]

Yes, this worked! now everythimg is working like it should. Thanks

closed 2020-11-10 by stale[bot]