Misiu/argon40 #3: I still cannot get I2C enabled.

Misiu, 2020-07-21:
I still cannot get I2C enabled.  I followed the instructions above and created the file rpi-i2c.conf in the /etc/modules-load.d folder and edited config.txt, but it still says it cannot set up Argon40.  When I went to edit the config.txt, I did not find a line that said #dtparam=i2c1=on.  So, I added it without the #.  I still get the following in the logs.  Any ideas?

`Logger: homeassistant.setup
Source: custom_components/argon40/__init__.py:64
First occurred: 11:41:33 PM (1 occurrences)
Last logged: 11:41:33 PM

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 34, in async_setup
    bus = SMBus(1)
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.7/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 193, in _async_setup_component
    result = await asyncio.wait_for(task, SLOW_SETUP_MAX_WAIT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/config/custom_components/argon40/__init__.py", line 64, in async_setup
    address,
UnboundLocalError: local variable 'address' referenced before assignment`

_Originally posted by @digitalMedic in https://github.com/Misiu/argon40/issues/1#issuecomment-661614210_

Misiu, 2020-07-21:
@digitalMedic please upvote my Feature request: https://community.home-assistant.io/t/allow-enabling-i2c-and-1wire-via-home-assistant-cli/208153

How did you create the file? Using File editor integration? I've tried that method couple of times during development and it worked every time.
Are you using Raspberry Pi 4 and Argon ONE case?

digitalMedic, 2020-07-21:
I used the built in HA file editor to create the file.  I have a RPI3B+ and an ArgonOne case for the RPI3B+.  I use my RPI4 and its ArgonONE case for something else.

Misiu, 2020-07-21:
@digitalMedic did you try [the official guide](https://www.home-assistant.io/hassio/enable_i2c/)?

Misiu, 2020-07-28:
@digitalMedic any updates?

digitalMedic, 2020-08-05:
I tried the official way to turn on I2C as well and it did not do anything.  I created the files as described and in HA I clicked on import from USB.  My USB drive light flickered briefly, but the system never rebooted.  I waited about 10 minutes and rebooted it myself and it did not work.  I am not sure why I cannot get this working.

webhostsg, 2020-08-09:
> I tried the official way to turn on I2C as well and it did not do anything. I created the files as described and in HA I clicked on import from USB. My USB drive light flickered briefly, but the [...]

Had the same issue, the official method doesn't seem to do anything. So i tried https://github.com/Misiu/argon40/issues/1#issuecomment-657207579, but i don't have any idea how to check if it's enabled.

Misiu, 2020-08-09:
Guys I can't do much about I2C.
Please create issues on HA repo and upvote and comment my feature request: https://community.home-assistant.io/t/allow-enabling-i2c-and-1wire-via-home-assistant-cli/208153

Misiu, 2020-08-09:
@digitalMedic please try this solution: https://github.com/Misiu/argon40/issues/4#issuecomment-671076667 created by @webhostsg

digitalMedic, 2020-08-15:
> Guys I can't do much about I2C.
> Please create issues on HA repo and upvote and comment my feature request: https://community.home-assistant.io/t/allow-enabling-i2c-and-1wire-via-home-assistant-cli/208153

No worries.  Someday it will get worked out.  I upvoted your feature request quite some time ago. :)

Misiu, 2020-09-21:
@digitalMedic there is an update in the HA docs: https://community.home-assistant.io/t/wth-why-enabling-i2c-is-so-hard/219771/12

https://www.home-assistant.io/hassio/enable_i2c/
Please try it and report back, so we can close this issue. Thanks!

nmarshall23, 2020-10-05:
The method in the HA docs worked for me.

Misiu, 2020-10-05:
@nmarshall23 thank you for confirming

digitalMedic, 2020-10-14:
I finally got it to work!

Misiu, 2020-10-14:
@digitalMedic can you share how you did it? Maybe others will find your method helpful.

digitalMedic, 2020-10-14:
I followed the instructions on the new HA docs.  Not sure what the difference was from before, but it worked this time.

closed 2020-11-20 by stale[bot]

sekt1953, 2021-07-15:
this her work for me:
## ENABLE I2C VIA HOME ASSISTANT OPERATING SYSTEM TERMINAL 
I what to enable i2c to bee able to controll fan speed i the Argon40 
### inspiration: 
* [ENABLE I2C VIA HOME ASSISTANT OPERATING SYSTEM TERMINAL](https://www.home-assistant.io/common-tasks/os#enable-i2c-via-home-assistant-operating-system-terminal)
* [Argon40Tech/Argon-ONE-i2c-Codes](https://github.com/Argon40Tech/Argon-ONE-i2c-Codes)
* [Misiu/argon40](https://github.com/Misiu/argon40)
* [HassOS I2C Configurator](https://community.home-assistant.io/t/add-on-hassos-i2c-configurator/264167)

You can enable i2c via this Raspberry Pi Console:

* Login as root.
  * Type login and press enter to access the shell.
* Type the following to enable i2c, you may need to replace sda1 with sdb1 or mmcblk0p1 depending on your platform:
```
mkdir /tmp/mnt
mount /dev/sda1 /tmp/mnt
mkdir -p /tmp/mnt/CONFIG/modules
echo -ne i2c-dev>/tmp/mnt/CONFIG/modules/rpi-i2c.conf
```
Open /tmp/mnt/config.txt with the editor "vi"
```
vi /tmp/mnt/config.txt
```
find the line in config.txt
```
#dtparam=i2c_arm=on
```
and change it to this
```
dtparam=i2c_arm=on
```
Save and exit vi with thise command:  
[ESC]:w  
[ESC]:x
```
sync
reboot
```

sekt1953, 2021-07-15:
You can fine my istallation description here:
https://github.com/mstedet/ESP32-2020#argon-one-m2---home-assistant-os-6x--supervised-version

Svenn-Erik