Misiu/argon40 #4: Invalid config

webhostsg, 2020-08-09:
## Version of the custom_component

`VERSION = "0.0.1"`
## Configuration

```yaml

argon40:

```

## Describe the bug
Followed the instructions copied the `argon40` folder to `custom_components`, added `argon40:` to `configuration.yaml`.

Reboot HA, received `Invalid config` notification when reboot completed.

## Debug log

```text

Log Details (ERROR)
Logger: homeassistant.setup
Source: custom_components/argon40/__init__.py:79
First occurred: 5:29:52 PM (1 occurrences)
Last logged: 5:29:52 PM

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 38, in async_setup
    bus = SMBus(1)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 193, in _async_setup_component
    result = await asyncio.wait_for(task, SLOW_SETUP_MAX_WAIT)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 483, in wait_for
    return fut.result()
  File "/config/custom_components/argon40/__init__.py", line 79, in async_setup
    address,
UnboundLocalError: local variable 'address' referenced before assignment
Home Assistant has started!

```

Misiu, 2020-08-09:
@webhostsg you must enable I2c on your Home Assistant.
Take a look at one of closed issues. There is a description.
I had the same problem.

webhostsg, 2020-08-09:
> @webhostsg you must enable I2c on your Home Assistant.
> Take a look at one of closed issues. There is a description.
> I had the same problem.

Thanks for the quick reply, I've tried the HA official USB method of enabling I2C doesn't seem to do anything, I've also tried your method, https://github.com/Misiu/argon40/issues/1#issuecomment-657207579, that log above was after doing the manual method of enabling I2C

Misiu, 2020-08-09:
Sadly there is nothing much I can do. I, created a feature request on HA forum. I've even looked into code. I can add the commands to CLI and Supervisor, but I don't have experience with OS development, so I can't to that part.
Create issue in official HA repo, maybe the core developers will help.
When I2C is enabled the component works perfect.

webhostsg, 2020-08-09:
> Sadly there is nothing much I can do. I, created a feature request on HA forum. I've even looked into code. I can add the commands to CLI and Supervisor, but I don't have experience with OS [...]

No issues, just sharing the issues so if someone has the same issues would know. Having the fan jumper to "Always on" on the Argon is really too noisy, so really hoping could make this work one day.. Thanks for all your effort.

webhostsg, 2020-08-09:
Got I2C enabled! I can hear the fan running now, haven't done any automation for it yet.

It's a mix between all these:
https://github.com/Misiu/argon40/issues/1#issuecomment-657207579
this:
https://www.home-assistant.io/hassio/enable_i2c/
and finally this:
https://github.com/home-assistant/operating-system/blob/ec6f7978b52823098c3d555f14c6097d0eb33916/Documentation/configuration.md#automatic

So this is what I did:
Format a USB stick with FAT32/EXT4/NTFS and name it `CONFIG` (in all capitals)

In the root of the USB drive add a folder called `modules`

Inside that folder add a text file called `rpi-i2c.conf` with the following contents:
```
i2c-bcm2708
i2c-dev
```
_If you are using a Windows machine make sure it does not save as `rpi-i2c.conf.txt`_

In the root of the USB drive add a file called `config.txt` with the following contents:
```
dtparam=i2c1=on
dtparam=i2c_arm=on
```
_If you are using a Windows machine make sure it does not save as `config.txt.txt`_

Insert the USB drive into your Raspberry Pi.
Now go to your Home Assistant web interface, in the sidebar click **Supervisor > System.**

Misiu, 2020-08-09:
@webhostsg thank you for sharing your solution. I have a free PI so I'll try it and add it to Readme.

webhostsg, 2020-08-09:
> @webhostsg thank you for sharing your solution. I have a free PI so I'll try it and add it to Readme.

you might want to add doing a host reboot with the USB plugged in might take around 10 minutes for it to completely boot up.

bussio, 2020-08-19:
I did it exactly as stated on @webhostsg. Unfortunately that doesn't work. I get the following error message:
[image]
What went wrong there?

kernelsandoz, 2020-08-25:
The node-red sequence looks nice. could you share the code ? Thanks

webhostsg, 2020-08-25:
> The node-red sequence looks nice. could you share the code ? Thanks

Sure!
[image]

```
[
    {
        "id": "6fc7831a.3df8bc",
        "type": "server-state-changed",
        "z": "7068ac6f.469094",
        "name": "CPU Temp",
        "server": "e29894fd.1104d8",
        "version": 1,
        "exposeToHomeAssistant": false,
        "haConfig": [
            {
                "property": "name",
                "value": ""
            },
            {
                "property": "icon",
                "value": ""
            }
        ],
        "entityidfilter": "sensor.cpu_temp",
        "entityidfiltertype": "exact",
        "outputinitially": true,
        "state_type": "num",
        "haltifstate": "1",
        "halt_if_type": "num",
        "halt_if_compare": "gte",
        "outputs": 2,
        "output_only_on_state_change": true,
        "x": 160,
        "y": 820,
        "wires": [
            [
                "41ae3fd6.2ec97"
            ],
            [
                "41ae3fd6.2ec97"
            ]
        ]
    },
    {
        "id": "41ae3fd6.2ec97",
        "type": "switch",
        "z": "7068ac6f.469094",
        "name": "CPU Temps",
        "property": "payload",
        "propertyType": "msg",
        "rules": [
            {
                "t": "btwn",
                "v": "0",
                "vt": "num",
                "v2": "44",
                "v2t": "num"
            },
            {
                "t": "btwn",
                "v": "44",
                "vt": "num",
                "v2": "46",
                "v2t": "num"
            },
            {
                "t": "btwn",
                "v": "46",
[... 284 lines cut ...]
        "debugenabled": true,
        "service_domain": "hassio",
        "service": "host_reboot",
        "entityId": "",
        "data": "",
        "dataType": "json",
        "mergecontext": "",
        "output_location": "payload",
        "output_location_type": "msg",
        "mustacheAltTags": false,
        "x": 1090,
        "y": 780,
        "wires": [
            []
        ]
    },
    {
        "id": "e29894fd.1104d8",
        "type": "server",
        "z": "",
        "name": "Home Assistant",
        "addon": true
    }
]
```

kernelsandoz, 2020-08-26:
Thanks for posting the flow. I have gone from "Hello World" to this.

I found I needed to add a couple of more lines to configuration.yaml

#Argon Pi Case integration - custom component argon40 does not load without this
argon40:

#sensor.cpu_temp sensor: not explicit in installation instructions, maybe expected knowledge.
sensor:

- platform: command_line
  name: CPU Temp
  command: "cat /sys/class/thermal/thermal_zone0/temp"
  unit_of_measurement: "°C"
  value_template: '{{value|multiply(0.001)|round(1)}}'

Tested the flow by reducing to 2 cases and setting temp around what it is now and can hear the fan turn on and off and see temp reported in node-red.

Next stop get a card in Lovelace.

Thanks again to Misiu and all who added comments (webhostsg)

webhostsg, 2020-08-26:
Just an update, today I tried changing from the 32 bit Raspberry Pi 4 image of Home Assistant to 64 bit, just wanted to test something out not related to Argon40.

I found out that the 64-bit version had issues with i2C, when it booted up the first time could not enable i2C, so I did my method that I had posted earlier and i2C was enabled and the fan started spinning. During my testing, I had to do several host reboot. After the second reboot I could not enable i2C anymore, tried doing the steps I usually do but to no avail. Good thing i had the 32 bit version on another SD card, just switch over and all working fine.

The 32 bit no issues so far, countless reboots.

Misiu, 2020-08-26:
@webhostsg please report such problems in Home Assistant OS repo. With more and more issues reported maybe they'll fix this issue once and for all.

Grey-Sheep, 2020-10-02:
> > The node-red sequence looks nice. could you share the code ? Thanks
> 
> Sure! But adjust the temps accordingly, where I stay is always warm and humid.
> 
> [image]
> 
> ```
> [
>     {
>         "id": "6fc7831a.3df8bc",
>         "type": "server-state-changed",
>         "z": "7068ac6f.469094",
>         "name": "CPU Temp",
>         "server": "e29894fd.1104d8",
>         "version": 1,
>         "exposeToHomeAssistant": false,
>         "haConfig": [
>             {
>                 "property": "name",
>                 "value": ""
>             },
>             {
>                 "property": "icon",
>                 "value": ""
>             }
>         ],
>         "entityidfilter": "sensor.cpu_temp",
>         "entityidfiltertype": "exact",

Many thanks for sharing the node red flow.

I had to make one small change which was to change the line from

"entityidfilter":"sensor.cpu_temp"

to 

"entityidfilter":"sensor.cpu_temperature"

closed 2020-11-08 by stale[bot]

EliotHertenstein, 2020-11-18:
> > The node-red sequence looks nice. could you share the code ? Thanks
> 
> Sure! But adjust the temps accordingly, where I stay is always warm and humid.
> 
> [image]
> 
> ```
> [
>     {
>         "id": "6fc7831a.3df8bc",
>         "type": "server-state-changed",
>         "z": "7068ac6f.469094",
>         "name": "CPU Temp",
>         "server": "e29894fd.1104d8",
>         "version": 1,
>         "exposeToHomeAssistant": false,
>         "haConfig": [
>             {
>                 "property": "name",
>                 "value": ""
>             },
>             {
>                 "property": "icon",
>                 "value": ""
>             }
>         ],
>         "entityidfilter": "sensor.cpu_temp",
>         "entityidfiltertype": "exact",
>         "outputinitially": true,
>         "state_type": "num",
>         "haltifstate": "1",
>         "halt_if_type": "num",
>         "halt_if_compare": "gte",
>         "outputs": 2,
>         "output_only_on_state_change": true,
>         "x": 160,
>         "y": 820,
>         "wires": [
>             [
>                 "41ae3fd6.2ec97"
>             ],
>             [
>                 "41ae3fd6.2ec97"
>             ]
>         ]
>     },
>     {
>         "id": "41ae3fd6.2ec97",
>         "type": "switch",
>         "z": "7068ac6f.469094",
>         "name": "CPU Temps",
>         "property": "payload",
>         "propertyType": "msg",
>         "rules": [
>             {
>                 "t": "btwn",
>                 "v": "0",
>                 "vt": "num",
>                 "v2": "44",
>                 "v2t": "num"
>             },
>             {
>                 "t": "btwn",
>                 "v": "44",
>                 "vt": "num",
>                 "v2": "46",
>                 "v2t": "num"
>             },
>             {
>                 "t": "btwn",
[... 287 lines cut ...]
>         "service": "host_reboot",
>         "entityId": "",
>         "data": "",
>         "dataType": "json",
>         "mergecontext": "",
>         "output_location": "payload",
>         "output_location_type": "msg",
>         "mustacheAltTags": false,
>         "x": 1090,
>         "y": 780,
>         "wires": [
>             []
>         ]
>     },
>     {
>         "id": "e29894fd.1104d8",
>         "type": "server",
>         "z": "",
>         "name": "Home Assistant",
>         "addon": true
>     }
> ]
> ```

Could you also send the Doubletap one? I can't figure it out for the life of me

sabbrowne, 2020-12-06:
> > The node-red sequence looks nice. could you share the code ? Thanks
> 
> Sure! But adjust the temps accordingly, where I stay is always warm and humid.
> 

Hi @webhostsg ,
I have been using this Node-Red flow for several months now... Its brilliant... Rock steady. I was wondering if you could modify this flow to include a HA Entity (type Sensor) to update a sensor with the Speed of the Fan that the Call Service node made using the argon integration.
Ie; when the Fan Speed 0 is called set the sensor to 0; or when the Fan Speed 30 is called set the sensor to 30; etc
I have tried myself but my skills are very low

Cheers

Shane