Misiu/argon40 #5: Cannot find Argon integration

doronsever, 2020-09-06:
## Version of the custom_component

## Configuration

```yaml

Add your logs here.

```

## Describe the bug
Hi,
I cannot find the Argon integration when try to search it.
I followed your steps here.
Do I need to do something else before that?
I started a fresh HA installation on RSP PI4 (64bit) and followed the instruction here.
I don't know what else I should bring you

## Debug log

```text

Add your logs here.

```

Misiu, 2020-09-07:
Hi @doronsever,

the integration currently doesn't have any UI.
If you followed the install steps correctly you should have two more services in Developer Tools > Services.

Misiu, 2020-09-07:
Also, did you add `argon40` to `configuration.yaml` as listed in step 8?

doronsever, 2020-09-10:
Hi
I see the 2 integtration, it seems that the fan is set the always_on but I'm not sure that it actually work.
I tried to set this flag to false and didn't hear any noise reduction
How can I actually know if it works or not?

Misiu, 2020-09-10:
Go to developer-tools/service and select `argon40.set_fan_speed`.
This service is responsible for setting fan speed.

doronsever, 2020-09-10:
Yes, when trying to change it, nothing happens...

Misiu, 2020-09-10:
see the logs (config/logs).
Maybe there's something that will help.

Also when selecting a service, there is a button that allows filling with example data.
Try doing that, change speed to 0, then to 100.

Misiu, 2020-09-14:
@doronsever any updates?
If the problem is solved please close this issue.

doronsever, 2020-09-23:
It seems to be working now :)
There is no UI integration

closed 2020-09-23 by doronsever

Misiu, 2020-09-23:
@doronsever you are correct, there is no UI yet.
I'm waiting for https://github.com/home-assistant/core/pull/39204
With that change, I'll be able to add UI for the fan and support 0-100% speed.

bussio, 2020-11-02:
If I bottle the image "https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-64-4.15.img.gz" with ballena etcher on the stick, then I find no way to follow the installation of "https://www.home-assistant.io/hassio/enable_i2c/". There is no Hassos-boot partition ... is there any other way to activate ic2 to make the Argon: integration work?

Misiu, 2020-11-02:
I only used SD card before, but IMO it should behave the same.
What operating system do You use?
You can always ask on HA forum for I2C enabled problems.
Recent versions of HA have the option to restore I2C settings via backup, so you can try that, but I'm not sure if I2C will work that way. Ref: https://github.com/home-assistant/supervisor/issues/1866

bussio, 2020-11-03:
I use Hassos_rpi4-64 4.15 on a SSD.

Misiu, 2020-11-03:
4.15 won't work on SSD. I think you need 5.4 (https://github.com/home-assistant/operating-system/releases).
I didn't have a chance to test running HA on SSD, so I can't help with that :/

bussio, 2020-11-03:
Excuse me, this is the hassos_rpi4-64-5.4.