Misiu/argon40 #7: Can the powerbutton used to shutdown the Pi with Home Assistant?

carsten-h, 2020-09-16:
Hello!

I only have a question regarding this integration.
Will this integration allow the use of the powerbutton to shutdown the Pi with Home Assistant?

Misiu, 2020-09-16:
@carsten-h I think it is possible.
Take a look at https://github.com/Misiu/argon40#bonus---button-double-tap-detection
it shows how to use double-tap event  to show notification.
You must change the service to `hassio.host_shutdown`.

Let me know it that worked for you.

I'm waiting for https://github.com/home-assistant/core/pull/39204, then I'll rewrite this integration, so it will be shown as a fan in HA and will try to support more button actions.

carsten-h, 2020-09-16:
Ah, OK, so you can only use the double tap in the moment.

Until now I don't have this enclosure for the Home Assistant Pi. I am thinking about buying it and your integration is a big step forward! :-)

I think the enclosure is great and I am already using it for a Pi4 for piCorePlayer with Logitech Media Server installed.

Misiu, 2020-09-16:
@carsten-h I would suggest to wait for next version of the case. Currently there are 3 version of the case: first one that doesn't support `always on`, second that has I2C command to turn on `always on` and third that has a number to turn on that feature.
Read more here: https://github.com/Argon40Tech/Argon-ONE-i2c-Codes/issues/1#issuecomment-691946555
This is needed for HA, because when you have a power loss you want you server to start when the power comes back.

Besides that the next version will support M2 NVMe SSD, so I guess it's worth waiting :)

Misiu, 2020-09-16:
> Ah, OK, so you can only use the double tap in the moment.

That's something with the rpi_gpio library, as I mentioned I'm planning to rewrite the integration and use gpiozero or different library to detect press, long press, and other actions, but I must wait for the creator's reply, because the MCU in the case has some actions hardcoded, for example, a long press will force shutdown the PI.

carsten-h, 2020-09-16:
> a long press will force shutdown the PI.

Yes, I know! Thank you! I will wait.

> This is needed for HA, because when you have a power loss you want you server to start when the power comes back.

No, not really. It should not switch on automatically. Sometimes there is a powerloss and shortly after the power is back again there is a new one. So the Pi and my NAS (both are attached to a UPS) are shutting down if there is no power for 10 minutes and then they stay switched off.

> Besides that the next version will support M2 NVMe SSD, so I guess it's worth waiting :)

That is a real reason to wait! Thank you!

Misiu, 2020-09-16:
Imagine that you have a UPS, it will help with short power looses, but if You have a longer power loss (let's say one hour) and the power is back the NAS will start automatically (that's how I configured mine), but your HA will not (if you use the default configured argon case). If you don't use the Argon case then it will start normally.
I was using the case for some time until the above issue. Now I have a version with a jumper and with always-on everything works just fine.

I'll let you know when the next version of my integration and the case will be released

carsten-h, 2020-09-16:
> I'll let you know when the next version of my integration and the case will be released

Thank you!

carsten-h, 2020-10-17:
I am using the normal Argon One case now for Home Assistant. Your scripts from here are working fine for the fan. I also can use the double press for shutting down Home Assistant. Great!
But I don‘t know how to automatically shut down the power of the case after shutting down Home Assistant. The shutdown is really taking long time, so it is not easy to see when it is safe to power off.

dieneuser, 2020-10-31:
> But I don‘t know how to automatically shut down the power of the case after shutting down Home Assistant. The shutdown is really taking long time, so it is not easy to see when it is safe to power off.

I also have the argon case and this integration, since yesterday and looking for a way to turn off/power cut the raspberry after HA has completely shut down.
In the installation script from argon (argon1.sh) I can see this functionality in line 106 by `bus.write_byte(0x1a,0xFF)`. 
@Misiu : Is it possible to add a function/service call (e.g argon40.power_cut) in this integration that does this?

THX

Misiu, 2020-11-02:
> In the installation script from argon (argon1.sh) I can see this functionality in line 106 by `bus.write_byte(0x1a,0xFF)`.
> @Misiu : Is it possible to add a function/service call (e.g argon40.power_cut) in this integration that does this?

@dieneuser I can add that, no problem, but if you call that during shutting down HA, it might cut power too soon.
I'll analyze the script once again and see how it behaves on Raspbian.

I'm waiting for the new fan integration in Home Assistant, when it will be implemented I'll be able to rewrite my custom component to be a fan with 0-100 speed control.

carsten-h, 2020-11-02:
> but if you call that during shutting down HA, it might cut power too soon.

That is clear. It has to be issued shortly before the Pi is down. I don't know how the scripts handle this.
I expect, that you have to have access to HassOS.

closed 2020-12-09 by stale[bot]