Misiu/argon40 #9: Argon40 Fan Speed Sensor

sabbrowne, 2020-09-27:
RPi4B 4gb
CPU 2100mhz
Hassos
Hi,
This integration works brilliantly, however I feel it needs a few enhancements if you have the time or if even possible:

1. Add a sensor that displays the Fan Speed in % of full speed. My reasoning behind this I have set up several automatons that change/control/proportionally the fan speed % according to CPU temperatures above 50 DEGC. Currently the only way to determine what speed the fan is at is by looking at most recent automation time stamp. I have considered creating automatons or monitor events when a particular automation is called but this would be cumbersome.

2. Currently when Home assistant is restarted the case fan automatically turns on to a predetermined speed (cant tell what speed, but sounds approx. 50%) Therefore an option to Set a boolean flag so that in configuration.yaml the fan speed can be set to off or a predetermined value. The fan speed will then be changed according to automation triggers. I have created an Automation to so that runs on start to set fan to of, but would be great to see this as a configuration option.

Thanks in advance or your consideration
Regards,
Shane

Misiu, 2020-09-28:
@sabbrowne thanks for using it and for the ideas.
You are right, the integration needs some work.
I'm waiting to core PR (https://github.com/home-assistant/core/pull/39204)https://github.com/home-assistant/core/pull/39204, then I'll be able to expose the fan in the case as a fan in the UI (with 0-100% control)
If I'm correct this will also solve your second problem/fr, because the state of the fan will be restored on startup, so it will be the same speed as before restart. If you need more control then I think I can add an option to set the initial speed.

When the Core PR get merged I'll start the rewrite.

sabbrowne, 2020-09-28:
> I'm waiting to core PR (home-assistant/core#39204, then I'll be able to expose the fan in the case as a fan in the UI (with 0-100% control)

Thanks. That’s brilliant. I’ll keep an eye open 

Cheers 

Shane

closed 2020-11-04 by stale[bot]

FreelancerJay, 2020-11-25:
> When the Core PR get merged I'll start the rewrite.

Sadly, the person working on the fan enhancements isn't able to work on it anymore (https://github.com/home-assistant/core/pull/39204#issuecomment-727198221).

If I had the knowhow, I'd be all over it, but I don't sadly. Hopefully someone picks up where they left off!

Misiu, 2020-11-25:
@FreelancerJay I saw that comment.
https://github.com/home-assistant/architecture/issues/127 take a look at this architecture issue.
Leave a comment there and on the forum. Maybe with bigger attention, someone will pick this.
I don't have that much experience with HA to move this forward.