Misiu/argon40 #14: init.py ERROR

fr33soul, 2020-11-19:
Logger: homeassistant.setup
Source: custom_components/argon40/__init__.py:79
First occurred: 21:53:16 (1 occurrences)
Last logged: 21:53:16

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 38, in async_setup
    bus = SMBus(1)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 213, in _async_setup_component
    result = await task
  File "/config/custom_components/argon40/__init__.py", line 79, in async_setup
    address,
UnboundLocalError: local variable 'address' referenced before assignment

Misiu, 2020-11-19:
@fr33soul it looks like you don't have I2C enabled.
Please follow this instructions: https://www.home-assistant.io/hassio/enable_i2c/

fr33soul, 2020-11-19:
> @fr33soul it looks like you don't have I2C enabled.
> Please follow this instructions: https://www.home-assistant.io/hassio/enable_i2c/

the instructions above didn't fix the issue.
After a research i found a solution for my Pi4 B with Argon40 case:
1. open the SD card with a Windows machine and edit the config.txt file
2. add the following 2 lines
dtparam=i2c1=on
dtparam=i2c_arm=on
3. insert back again the SD card into the raspberry Pi4 B and start
4. with SSH enabled, open the file located \etc\modules-load.d\rpi-i2c.conf
if doesn't exists, create it and add 2 lines:
i2c-bcm2708
i2c-dev
5. restart the raspberry Pi4 B

cdbailey, 2020-11-29:
My fix for this issue with a Pi4 B and Argon40 case had a few variations because I utilize the Argon40 M.2 case with built-in SATA M.2 storage running HASS 5.5 (currently development track).

1. Remove the USB port bridge and attach USB cable between the storage port of the Argon40 case and another system running Linux.
2. Mount the first partition (the FAT partition), in my case it was /dev/sdb1.
3. Edit the config.txt using either nano or vi, adding the following lines.

dtparam=i2c_vc=on
dtparam=i2c_dev=on

4. Create a directory titled CONFIG on the root of the drive.
5. Create a subdirectory under CONFIG called modules.
6. Create a file within the modules directory called rpi-i2c.conf with the contents of the two lines below.

i2c-dev
i2c-bcm2835

7. Unmount the drive and reinstall the USB port bridge.
8. Boot the Pi and wait five minutes, reboot to make sure the changes are activated.

closed 2021-01-06 by stale[bot]