Misiu/argon40 #17: [Bug] Fan Speed not changing

greghesp, 2021-01-15:
I've got an automation that turns the fan off if its below 50.  I can see that the automation has triggered and the CPU temperature is 30, however the fan speed does not change?

```
alias: '[Argon One] Turn Off Fan Below 50'
description: ''
trigger:
  - platform: numeric_state
    entity_id: sensor.cpu_temperature
    below: '50'
condition: []
action:
  - service: argon40.set_fan_speed
    data:
      speed: 0
mode: single
```

Misiu, 2021-01-15:
@greghesp can you try calling the service by hand? via the dev tools?
There will be sample data, just adjust the value.

greghesp, 2021-01-15:
@Misiu Hmm, it seems the service doesn't even exist

I've added the custom component:
[image]

Added argon40 to my config
```
argon40:
```

The only thing is that Argon40 doesn't show up under the Integration search?

greghesp, 2021-01-15:
Ah, just found this in the logs

```
Logger: homeassistant.setup
Source: custom_components/argon40/__init__.py:79
First occurred: 14:51:56 (1 occurrences)
Last logged: 14:51:56

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 38, in async_setup
    bus = SMBus(1)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 213, in _async_setup_component
    result = await task
  File "/config/custom_components/argon40/__init__.py", line 79, in async_setup
    address,
UnboundLocalError: local variable 'address' referenced before assignment
```

Misiu, 2021-01-15:
@greghesp you don't have I2C enabled.
Try this tool: https://community.home-assistant.io/t/argon-one-rpi-4-case-script/198719/29?u=misiu or use the official guide.

greghesp, 2021-01-15:
Ah bloody hell, needed a hard power down.  Was doing a HA restart lol

closed 2021-01-15 by greghesp

Misiu, 2021-01-15:
@greghesp the important part is that it works :)