Misiu/argon40 #19: i2c not enabled?

mgiebels, 2021-01-19:
I get a error on argon4

i have a raspberry pi4band i cant get enabled i2c?

some help?

Logger: homeassistant.setup
Source: custom_components/argon40/__init__.py:64
First occurred: 21:01:21 (1 occurrences)
Last logged: 21:01:21

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 34, in async_setup
    bus = SMBus(1)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.8/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 213, in _async_setup_component
    result = await task
  File "/config/custom_components/argon40/__init__.py", line 64, in async_setup
    address,
UnboundLocalError: local variable 'address' referenced before assignment

mcflexus, 2021-01-19:
You have to enable i2c yourself. Easy way: [thru this addon](https://community.home-assistant.io/t/add-on-hassos-i2c-configurator/264167) . little less easy way, google enable i2c hassio.

mgiebels, 2021-01-19:
a that was easy! thank you!

closed 2021-02-26 by stale[bot]