Misiu/argon40 #20: No 'version' key in the manifest file - breaking change in upcoming Home Assistant release

crgbt, 2021-04-07:
Current release results in the following message being logged in the Home Assistant core logs (Configuration > Logs):

```
2021-04-07 21:28:57 WARNING (MainThread) [homeassistant.loader] No 'version' key in the manifest file for custom integration 'argon40'. This will not be allowed in a future version of Home Assistant. Please report this to the maintainer of 'argon40'
```

I've checked `manifest.json` and there is already a version key present, however for some reason it is not being recognised.

More details can be found here: https://www.home-assistant.io/blog/2021/03/03/release-20213/#breaking-changes and here: https://developers.home-assistant.io/docs/creating_integration_manifest/

Misiu, 2021-04-07:
Hi there,
I've recently added the version into manifest file, but it isn't released yet as a new version. I'll do that soon 🙂

Misiu, 2021-04-10:
@crgbt second release released. Please try it and report back if you have any problems.

crgbt, 2021-04-11:
The new version clears the log file entry - thanks for fixing this @Misiu. It appears that the new version doesn't seem to like setting the fan speed to 0% though, shall I open a separate issue to report this?

Misiu, 2021-04-11:
Yes, please create a separate issue for that (I didn't change the logic for 0%, the previous version worked fine?)
I have a free Pi4 and argon case so in my free time I'll create a dev environment and give this a try.

Any errors in logs?

Misiu, 2021-04-29:
The issue with manifest is fixed.

closed 2021-04-29 by Misiu