Misiu/argon40 #21: Integration failed to initialize.

greghesp, 2021-04-10:
## Version of the custom_component
0.0.2

## Configuration

```yaml

argon40:

```

## Describe the bug
Argon 40 does not appear in Integration list after.  I have redownloaded it from Github, copied it over, added to my configuration.yaml, restarted but it still isn't in the integration list.  Have restarted several times

## Debug log

```text

Argon40
Version: 0.0.2
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/Misiu/argon40/issues
-------------------------------------------------------------------
2021-04-10 17:07:24 INFO (MainThread) [homeassistant.setup] Setup of domain argon40 took 0.0 seconds
2021-04-10 17:07:24 ERROR (MainThread) [homeassistant.setup] Setup failed for argon40: Integration failed to initialize.
```

Misiu, 2021-04-10:
@greghesp it won't be on the integration list because it doesn't have config flow.
Do you have I2C enabled?

greghesp, 2021-04-10:
> @greghesp it won't be on the integration list because it doesn't have config flow.
> Do you have I2C enabled?

I think you've just jogged my memory.  I've just flashed the bios and reinstalled on an SSD!  Will report back!

Misiu, 2021-04-10:
@greghesp I hope everything will work fine.
I plan to do a full rewrite to include config flow and I2C detection, but sadly not much time lately.

greghesp, 2021-04-10:
@Misiu I think that was the issue. I2C add-on detected it was disabled, so rebooting now

greghesp, 2021-04-10:
Yep, that fixed it. Thanks!

closed 2021-04-10 by greghesp