Misiu/argon40 #22: Fan control for 'speed 0' not working in release 2

crgbt, 2021-04-11:
Hi @Misiu, since release 2 it appears that setting the fan speed to 0% does not work. I have tried re-creating the automations and re-installing the component files, and I've not been able to identify anything wrong in the logs. My hardware is a Raspberry Pi 4 8GB running the Argon ONE M.2 enclosure.

I've redacted location data from the log entries for privacy, however the remaining entries are intact. If there are any additional configuration files or log entries that will be of assistance please let me know and I'll supply them for you.

Fan speed is being detected correctly; currently at 38.9 °C. I'm unable to confirm at which speed the fan is running based on my automation entries, but it is spinning.

## Logs
<details><summary>HA debug logs refrencing term 'argon40'</summary>

```
2021-04-11 16:15:24 WARNING (MainThread) [homeassistant.loader] You are using a custom integration argon40 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-11 16:15:27 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'dhcp', 'media_source', 'rpi_power', 'timer', 'input_boolean', 'tag', 'sensor', 'map', 'tado', 'hue', 'default_config', 'sun', 'zone', 'history', 'input_select', 'met', 'updater', 'group', 'input_number', 'homekit_controller', 'my', 'mobile_app', 'ssdp', 'tts', 'sonos', 'automation', 'logbook', 'input_text', 'synology_dsm', 'argon40', 'deconz', 'counter', 'scene', 'input_datetime', 'zwave_js', 'system_health', 'script', 'zeroconf', 'blueprint', 'roku', 'cast'}
2021-04-11 16:15:29 INFO (MainThread) [homeassistant.setup] Setting up argon40
2021-04-11 16:15:29 INFO (MainThread) [custom_components.argon40]
-------------------------------------------------------------------
Argon40
Version: 0.0.2
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/Misiu/argon40/issues
-------------------------------------------------------------------
2021-04-11 16:15:29 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event service_registered[L]: domain=argon40, service=set_fan_speed>
2021-04-11 16:15:29 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event service_registered[L]: domain=argon40, service=set_mode>
2021-04-11 16:15:29 INFO (MainThread) [homeassistant.setup] Setup of domain argon40 took 0.0 seconds
2021-04-11 16:15:29 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=argon40>
2021-04-11 16:15:31 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [548198226192] Sending {'id': 25, 'type': 'result', 'success': True, 'result': {'latitude': REDACTED, 'longitude': REDACTED, 'elevation': REDACTED, 'unit_system': {'length': 'km', 'mass': 'g', 'pressure': 'Pa', 'temperature': '°C', 'volume': 'L'}, 'location_name': 'Home', 'time_zone': 'Europe/London', 'components': {'image', 'binary_sensor.rpi_power', 'dhcp', 'media_source', 'rpi_power', 'timer', 'logger', 'input_boolean', 'binary_sensor.updater', 'auth', 'tag', 'sensor', 'media_player.sonos', 'system_log', 'map', 'onboarding', 'cloud', 'persistent_notification', 'sensor.command_line', 'sun', 'alexa', 'zone', 'history', 'input_select', 'homeassistant', 'updater', 'group', 'input_number', 'binary_sensor', 'frontend', 'my', 'ssdp', 'config', 'api', 'tts', 'media_player', 'sonos', 'http', 'logbook', 'person', 'recorder', 'input_text', 'argon40', 'counter', 'scene', 'input_datetime', 'zwave_js', 'device_automation', 'search', 'system_health', 'script', 'lovelace', 'hassio', 'blueprint', 'scene.homeassistant', 'webhook', 'zeroconf', 'stt', 'websocket_api'}, 'config_dir': '/config', 'whitelist_external_dirs': {'/config/www', '/media'}, 'allowlist_external_dirs': {'/config/www', '/media'}, 'allowlist_external_urls': set(), 'version': '2021.3.4', 'config_source': 'storage', 'safe_mode': False, 'state': 'NOT_RUNNING', 'external_url': None, 'internal_url': None}}
2021-04-11 16:15:31 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [548198226192] Sending {'id': 32, 'type': 'result', 'success': True, 'result': {'latitude': REDACTED, 'longitude': REDACTED, 'elevation': REDACTED, 'unit_system': {'length': 'km', 'mass': 'g', 'pressure': 'Pa', 'temperature': '°C', 'volume': 'L'}, 'location_name': 'Home', 'time_zone': 'Europe/London', 'components': {'image', 'binary_sensor.rpi_power', 'dhcp', 'media_source', 'rpi_power', 'timer', 'logger', 'input_boolean', 'binary_sensor.updater', 'auth', 'tag', 'sensor', 'media_player.sonos', 'system_log', 'map', 'onboarding', 'cloud', 'persistent_notification', 'sensor.command_line', 'sun', 'alexa', 'zone', 'history', 'input_select', 'homeassistant', 'updater', 'group', 'input_number', 'binary_sensor', 'frontend', 'my', 'ssdp', 'config', 'api', 'tts', 'media_player', 'sonos', 'http', 'logbook', 'person', 'recorder', 'input_text', 'argon40', 'counter', 'scene', 'input_datetime', 'zwave_js', 'device_automation', 'search', 'system_health', 'script', 'lovelace', 'hassio', 'blueprint', 'scene.homeassistant', 'webhook', 'zeroconf', 'stt', 'websocket_api'}, 'config_dir': '/config', 'whitelist_external_dirs': {'/config/www', '/media'}, 'allowlist_external_dirs': {'/config/www', '/media'}, 'allowlist_external_urls': set(), 'version': '2021.3.4', 'config_source': 'storage', 'safe_mode': False, 'state': 'NOT_RUNNING', 'external_url': None, 'internal_url': None}}
2021-04-11 16:15:31 DEBUG (MainThread) [homeassistant.helpers.translation] Cache miss for en-GB: image, binary_sensor.rpi_power, dhcp, media_source, rpi_power, timer, logger, input_boolean, binary_sensor.updater, auth, tag, sensor, media_player.sonos, system_log, map, onboarding, cloud, persistent_notification, sensor.command_line, sun, alexa, zone, history, input_select, homeassistant, updater, group, input_number, binary_sensor, frontend, my, ssdp, config, api, tts, media_player, sonos, http, logbook, person, recorder, input_text, argon40, counter, scene, input_datetime, zwave_js, device_automation, search, system_health, script, lovelace, hassio, blueprint, scene.homeassistant, webhook, zeroconf, stt, websocket_api
2021-04-11 16:15:32 DEBUG (SyncWorker_7) [homeassistant.util.json] JSON file not found: /config/custom_components/argon40/translations/en-GB.json
2021-04-11 16:15:32 DEBUG (SyncWorker_0) [homeassistant.util.json] JSON file not found: /config/custom_components/argon40/translations/en.json
2021-04-11 16:15:34 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [548198226192] Sending {'id': 34, 'type': 'result', 'success': True, 'result': {'resources': {'component.websocket_api.title': 'Home Assistant WebSocket API', 'component.blueprint.title': 'Blueprint', 'component.stt.title': 'Speech-to-Text (STT)', 'component.zeroconf.title': 'Zero-configuration networking (zeroconf)', 'component.hassio.title': 'Home Assistant Supervisor', 'component.lovelace.title': 'Lovelace', 'component.script.title': 'Scripts', 'component.system_health.title': 'System Health', 'component.search.title': 'Search', 'component.device_automation.title': 'Device Automation', 'component.zwave_js.title': 'Z-Wave JS', 'component.input_datetime.title': 'Input Datetime', 'component.scene.title': 'Scenes', 'component.counter.title': 'Counter', 'component.argon40.title': 'Argon40', 'component.input_text.title': 'Input Text', 'component.recorder.title': 'Recorder', 'component.person.title': 'Person', 'component.logbook.title': 'Logbook', 'component.http.title': 'HTTP', 'component.sonos.title': 'Sonos', 'component.media_player.title': 'Media Player', 'component.tts.title': 'Text-to-Speech (TTS)', 'component.api.title': 'Home Assistant API', 'component.config.title': 'Configuration', 'component.ssdp.title': 'Simple Service Discovery Protocol (SSDP)', 'component.my.title': 'My Home Assistant', 'component.frontend.title': 'Home Assistant Frontend', 'component.binary_sensor.title': 'Binary Sensor', 'component.input_number.title': 'Input Number', 'component.group.title': 'Group', 'component.updater.title': 'Updater', 'component.homeassistant.title': 'Home Assistant Core Integration', 'component.input_select.title': 'Input Select', 'component.history.title': 'History', 'component.zone.title': 'Zone', 'component.webhook.title': 'Webhook', 'component.alexa.title': 'Amazon Alexa', 'component.sun.title': 'Sun', 'component.persistent_notification.title': 'Persiste [... cut ...]
supervisor_version | supervisor-2021.03.9
docker_version | 19.03.13
disk_total | 219.4 GB
disk_used | 4.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Let's Encrypt (4.11.0), NGINX Home Assistant SSL proxy (3.0.1), Samba share (9.3.1), File editor (5.2.0), deCONZ (6.8.0), MariaDB (2.3.0), Terminal & SSH (9.1.0), Z-Wave JS to MQTT (0.11.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

closed 2021-07-18 by stale[bot]

miamilabs, 2021-12-01:
Did you find a solution on this?
I am facing same issue. The Animation works if you trigger it by yourself, but after reboot the fan runs and wont turn off.

Misiu, 2021-12-02:
@miamilabs that's because I set the initial speed to 10.
The automation should set the speed to 0 but after some time.
Try adding automation from this issue: https://github.com/Misiu/argon40/issues/17#issue-786945245
Or you can set the speed to 0 when the system start: https://www.home-assistant.io/docs/automation/trigger/#home-assistant-trigger