Misiu/argon40 #23: Issue or Feature Agron One PI4 V2 FAN

lars-dev, 2021-04-12:
## Version of the custom_component
0.02
-->

## Configuration

```yaml

Add your logs here.

```

## Describe the bug
The fan of the ARGON ONE PI 4 V2 Case does not seem to be affected by this tool. I am not sure though if its a problem on my side or it is not supported yet.

## Debug log

```text

Add your logs here.

```

Misiu, 2021-04-12:
Did you follow the instructions?
Especially step 8 : `Add argon40: to your configuration.yaml`
You may have I2C enabled, but this should show some errors in the log.
Please check that and report back.

lars-dev, 2021-04-12:
I have figured it out now. I hadn't done anything to connect to the i2c 
Not it works.
Thanks so much for that tool. The high pitched fan noise was driving me crazy

closed 2021-04-12 by lars-dev