Misiu/argon40 #25: Argon 40 Won't Load

Originalme, 2021-05-18:
## Version of the custom_component

- Argon40 Version: 0.0.2

## Configuration

<details>
<summary>configuration.yaml</summary>

```yaml
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

argon40:

sensor:
  - platform: command_line
    name: CPU Temp
    command: "cat /sys/class/thermal/thermal_zone0/temp"
    unit_of_measurement: "°C"
    value_template: "{{ value | multiply(0.001) | round(1) }}"
    
logger:
  default: info
  logs:
    homeassistant.components.yamaha: critical
    custom_components.my_integration: debug
    homeassistant.setup: debug

```

</details>

## Description
Hello @Misiu, I hope you are doing well, and thank you for maintaining this extension.

I followed the instructions, and confirmed that i2c is functioning and reporting. I created the **custom_components/argon40** directory and uploaded the files, and added **argon40:** to my configuration.yaml. However, when restarting the server the plugin errors out when loading and fails to initialize.

I would be happy to provide any other data, and I apologize if I missed something. I only dabble with development, and I don't do any hardware stuff.

#### Version Info
- Machine model: Raspberry Pi 4 Model B Rev 1.4
- HassOS Image: 5.13
- Linux version: 5.4.83-v8
- Supervisor:  2021.04.3
- Core: 2021.5.4
- Argon40 Version: 0.0.2

## Debug log

```log
2021-05-18 11:57:00 INFO (MainThread) [homeassistant.setup] Setting up argon40
2021-05-18 11:57:00 INFO (MainThread) [custom_components.argon40]
-------------------------------------------------------------------
Argon40
Version: 0.0.2
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/Misiu/argon40/issues
-------------------------------------------------------------------
2021-05-18 11:57:00 INFO (MainThread) [homeassistant.setup] Setup of domain argon40 took 0.0 seconds
2021-05-18 11:57:00 ERROR (MainThread) [homeassistant.setup] Setup failed for argon40: Integration failed to initialize.
2021-05-18 11:57:00 ERROR (Thread-2) [root] Uncaught thread exception
Traceback (most recent call last):
File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
self.run()
File "/usr/local/lib/python3.8/threading.py", line 870, in run
self._target(*self._args, **self._kwargs)
File "/usr/local/lib/python3.8/logging/handlers.py", line 1487, in _monitor
self.handle(record)
File "/usr/local/lib/python3.8/logging/handlers.py", line 1468, in handle
handler.handle(record)
File "/usr/local/lib/python3.8/logging/__init__.py", line 954, in handle
self.emit(record)
File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 192, in emit
entry = LogEntry(record, stack, _figure_out_source(record, stack, self.hass))
File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 99, in __init__
self.message = deque([record.getMessage()], maxlen=5)
File "/usr/local/lib/python3.8/logging/__init__.py", line 373, in getMessage
msg = msg % self.args
```

Tedjuh, 2021-05-22:
have also same problem

Yeti-diver, 2021-05-23:
same behavior as described by Originalme:

**debug log**
2021-05-22 10:11:58 INFO (MainThread) [homeassistant.setup] Setting up argon40
2021-05-22 10:11:58 INFO (MainThread) [custom_components.argon40]

Argon40
Version: 0.0.2
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/Misiu/argon40/issues

2021-05-22 10:11:58 INFO (MainThread) [homeassistant.setup] Setup of domain argon40 took 0.0 seconds
2021-05-22 10:11:58 ERROR (MainThread) [homeassistant.setup] Setup failed for argon40: Integration failed to initialize.
2021-05-22 10:11:58 ERROR (Thread-2) [root] Uncaught thread exception
Traceback (most recent call last):
File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
self.run()
File "/usr/local/lib/python3.8/threading.py", line 870, in run
self._target(*self._args, **self._kwargs)
File "/usr/local/lib/python3.8/logging/handlers.py", line 1487, in _monitor
self.handle(record)
File "/usr/local/lib/python3.8/logging/handlers.py", line 1468, in handle
handler.handle(record)
File "/usr/local/lib/python3.8/logging/__init__.py", line 954, in handle
self.emit(record)
File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 192, in emit
entry = LogEntry(record, stack, _figure_out_source(record, stack, self.hass))
File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 99, in __init__
self.message = deque([record.getMessage()], maxlen=5)
File "/usr/local/lib/python3.8/logging/__init__.py", line 373, in getMessage
msg = msg % self.args
TypeError: %d format: a number is required, not NoneType

vlfesko, 2021-05-31:
Hi @Originalme @Tedjuh  @Yeti-diver
Had the same issue, it was caused by NOT enabled i2c on my RPi 4 device. To solve it you need to enable i2c in HassOS:
- [Official way](https://www.home-assistant.io/common-tasks/os#enable-i2c-via-home-assistant-operating-system-terminal)
- [Using Add-on HassOS I2C Configurator](https://community.home-assistant.io/t/add-on-hassos-i2c-configurator/264167)

Reboot few times (the addon-way requires two times, at first it put files, at second the config is applied), and then fan control will work.

I found myself that [this Argon One Active Cooling Addon](https://community.home-assistant.io/t/add-on-hassos-i2c-configurator/264167) easier to use if you don't need HACS.

Cheers, 🍷

Ukchris79, 2021-06-04:
> Hi @Originalme @Tedjuh @Yeti-diver
> Had the same issue, it was caused by NOT enabled i2c on my RPi 4 device. To solve it you need to enable i2c in HassOS:
>
> * [Official [...]

So I am still having issues with "Package argon40 setup failed. Component argon40 Integration 'argon40' not found." even after following your advice.

Am i the only one?

Yeti-diver, 2021-06-04:
Hi @all,

got Argon40 Cooling running. 

@vlfesko: 
Your advice "NOT enabled i2c on my RPi 4" was my trigger. 

Ukchris79:
I've connected my ssd (from pi) to my laptop and mount "hassos-boot". Made at first folder "CONFIG", than CONFIG/modules. In this folder i added rpi-i2c.conf with with the following content: 
 i2c-dev
Check config.txt if 
 dtparam=i2c_vc=on
 dtparam=i2c_arm=on
are inside.
Double time power off, thats it.

vlfesko, 2021-06-08:
> So I am still having issues with "Package argon40 setup failed. Component argon40 Integration 'argon40' not found." even after following your advice.
>
> Am i the only one?

Hi,
It's different issue, most likely related to not properly installed argon40 addon - your HA does not see the installed custom component, maybe `custom_components` folder is not within HA `config` folder.

ArjenWols, 2021-07-02:
Help?
Any advise or good tip is welcome<G>,

Custom component still not found after following the directions for adding Custom component by 
- Configuration, Integrations, + sign to Add integration (nothing with 'arg' found). 
core-2021.6.6
supervisor-2021.06.6
Home Assistant OS 6.1
Deployment: Production

The custom files are here:
[image]

My i2C log tells me:
crw-rw----    1 root     netdev     81,   3 Feb  2 16:29 video16
crw-------    1 root     root       10, 130 Feb  2 16:29 watchdog
crw-------    1 root     root      249,   0 Feb  2 16:29 watchdog0
crw-rw-rw-    1 root     root        1,   5 Feb  2 16:29 zero
brw-rw----    1 root     disk      254,   0 Feb  2 16:29 zram0
brw-rw----    1 root     disk      254,   1 Feb  2 16:29 zram1
brw-rw----    1 root     disk      254,   2 Feb  2 16:29 zram2
/dev/i2c-1
/dev/i2c-0 /dev/i2c-1 /dev/i2c-10 /dev/i2c-11
Found i2c access!  Nothing to do!  You can remove this add-on.

Misiu, 2021-07-05:
@ArjenWols how are you installing the integration?
Currently, it isn't possible to install integration using the + sign, you must add it via `configuration.yaml` - https://github.com/Misiu/argon40#installation
I'm working on a new version, but sadly I have very limited time.

ArjenWols, 2021-07-05:
> @ArjenWols how are you installing the integration?
> Currently, it isn't possible to install integration using the + sign, you must add it via `configuration.yaml` - https://github.com/Misiu/argon40#installation
> I'm working on a new version, but sadly I have very limited time.

Hello Misiu,
Thanks for the reply.
Installed HASS as a possible replacement for Domoticz.
Deleted the HASS after a few days of frustration because the MQTT integration of ESP8266 sensor devices did not work.
Will try this later, but not very soon.
Will keep your update in mind.
Thanks and have a nice day.

sekt1953, 2021-07-14:
I have to enable i2c on home assistant os befor start of installation 
I follow this link https://www.home-assistant.io/common-tasks/os#enable-i2c-via-home-assistant-operating-system-terminal

but it du not work all the way, so from console I did this

mkdir /tmp/mnt
mount /dev/sda1 /tmp/mnt
cd /tmp/mnt
vi config.txt

found the chapter:
#Uncomment some or all of thise to enable the optinal hardware interface
#dtparm=i2c_arm=no

and change it to 
# Uncomment some or all of thise to enable the optinal hardware interface
dtparm=i2c_arm=no

and the I rebooted and make the installation og Argon40

sekt1953, 2021-07-14:
I have the same problem, I came around it by folling this link: https://www.home-assistant.io/common-tasks/os#enable-i2c-via-home-assistant-operating-system-terminal

But this part did not work for me:

echo dtparam=i2c_vc=on >> /tmp/mnt/CONFIG/config.txt
echo dtparam=i2c_arm=on >> /tmp/mnt/CONFIG/config.txt

so I did this

Login again:

mkdir /tmp/mnt
mount /dev/sda1 /tmp/mnt
cd /tmp/mnt
vi /tmp/mnt/config.txt

and removed # from the line 
#dtparam=i2c_arm=on

and exit vi with
[ESC]:w
[ESC]:x

and rebooted, and now I can du the installation of Aron40

closed 2021-08-20 by stale[bot]