Misiu/argon40 #26: HomeAssistant Container

cargnellogiacomo, 2021-06-02:
## Version of the custom_component
latest version

## Configuration

argon40:

## Describe the bug
HomeAssistant installed as a container on docker on a RasperryPi 4 (4GB).
All argon40 files have been added in the custom components folder.
When I add the "argon40:" text in the configuration.yaml of HomeAssistant and I do the configuration check I receive an error (see below) and if I restart HA, the argon40 integration is not loaded.
The error states that the module should be used on a RaspberryPi while actually it is.

Do I perhaps need to forward some devices/port in the docker configuration?

## Debug log

Logger: aiohttp.server
Source: custom_components/argon40/__init__.py:5
Integration: argon40 (documentation, issues)
First occurred: 00:19:49 (1 occurrences)
Last logged: 00:19:49

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 422, in _handle_request
    resp = await self._request_handler(request)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_app.py", line 499, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_middlewares.py", line 119, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 56, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 18, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 74, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 135, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 131, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/core.py", line 29, in post
    errors = await async_check_ha_config_file(request.app["hass"])
  File "/usr/src/homeassistant/homeassistant/config.py", line 904, in async_check_ha_config_file
    res = await check_config.async_check_ha_config_file(hass)
  File "/usr/src/homeassistant/homeassistant/helpers/check_config.py", line 133, in async_check_ha_config_file
    component = integration.get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 485, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/custom_components/argon40/__init__.py", line 5, in <module>
    from RPi import GPIO  # pylint: disable=import-error
  File "/usr/local/lib/python3.8/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

elchanly, 2021-06-24:
+1, never could make it work

Misiu, 2021-06-24:
@cargnellogiacomo I'm not sure what must be changed to allow this to work with docker. I don't have much experience with docker.
The way the integration works is very simple - you must have I2C enabled and a device on a specific I2C address.
Probably Home Assistant on docker doesn't have access to I2C.
Take a look at this Stackoverflow answer: https://stackoverflow.com/a/40266284/965722, maybe this will help.
Please let me know if you have any success. I'll be more than happy to add a proper description to the readme and installation steps.

cargnellogiacomo, 2021-06-24:
let me do some tests in the coming days and I will let you know!

Misiu, 2021-07-23:
@cargnellogiacomo any updates?

cargnellogiacomo, 2021-07-25:
no update. I am using the Argon40 script installed directly on the RaspOS. No possible to manage via HomeAssistant but at least it controls the fan also in case HA crashes

closed 2021-08-31 by stale[bot]