Misiu/argon40 #28: compatibility with homeassistant 2021.8.6

s4-crashpc, 2021-08-12:
Hello,
after updating to version 2021.8.6 the component is no longer recognized, below the error;

Invalid config
The following integrations and platforms could not be set up:
argon40

Is it possible for you to update it to make it compatible?
Thanks in advance

Misiu, 2021-08-13:
Hi, I'm finishing a rewrite and will release it in a week or so. I've tested everything on 2021.7 and had no problem.
I'll keep you updated :)

s4-crashpc, 2021-08-13:
Thank you so much.

yiskang, 2021-08-30:
Hi, I'm not sure if it's related to this issue. Please kindly help me check. Thank you in advance!

I'm trying to install this addon with 2021.8.5, and followed the instruction under [Raspberry Pi 4, Home Assistant OS (5.5, dev version) on a SSD, and the Argon One M.2 Case (In Progress)](https://community.home-assistant.io/t/raspberry-pi-4-home-assistant-os-5-5-dev-version-on-a-ssd-and-the-argon-one-m-2-case-in-progress/248025), but saw these error messages under HassOS Logs. I typed `lsmod | grep i2c` in the host console. Looks everything is good. I can see i2c modules are loaded.  So, I'm wondering if it's v0.04 is compatible with 2021.8.5.

```
2021-xx-xx hh:mm:ss INFO (MainThread) [custom_components.argon40]
-------------------------------------------------------------------
Argon40
Version: 0.0.4
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/Misiu/argon40/issues
-------------------------------------------------------------------
2021-xx-xx hh:mm:ss INFO (MainThread) [homeassistant.setup] Setup of domain argon40 took 0.0 seconds
2021-xx-xx hh:mm:ss ERROR (MainThread) [homeassistant.setup] Setup failed for argon40: Integration failed to initialize.
```

maobuz, 2021-09-09:
Hi argon40 team,
i'm trying yo install your component but, following the installation guide, i can't find the service argon40.set_fan_speed

Can you check compatibility with version | core-2021.9.4 (and.5) ?

Please let me know, thank you.

maobuz, 2021-09-14:
@Misiu Please give me a news :P

theCrazylegs, 2021-09-30:
@maobuz me too :)

maobuz, 2021-09-30:
@theCrazylegs but @Misiu  not responding ..... :(

Misiu, 2021-09-30:
sorry for the late reply, but my son got sick some time ago and it's hard to find time to fix this.
I have a branch that supports config flow and it should fix this problem (I've also removed some unneeded code).
I'll try to push that as soon as possible, but as I wrote it's hard to find the time right now.

maobuz, 2021-09-30:
@Misiu i'm so sorry for your son.
do not worry, i'm stay tuned for update in the future. thanks !

C0pywriting, 2021-10-05:
Maybe logs will help here. These show up for me. 

I hope the son is already feeling better.

Version | core-2021.9.7
-- | --
Installationstyp | Home Assistant OS
Entwicklung | false
Supervisor | true
Docker | true
Benutzer | root
Virtuelle Umgebung | false
Python-Version | 3.9.7
Betriebssystemfamilie | Linux
Betriebssystem-Version | 5.10.17-v8
CPU-Architektur | aarch64

Host-Betriebssystem | Home Assistant OS 6.4
-- | --
Update-Channel | stable
Supervisor-Version | supervisor-2021.09.6
Docker-Version | 20.10.7

`2021-10-05 17:02:46 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-10-05 17:02:46 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration argon40 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-10-05 17:02:50 ERROR (MainThread) [homeassistant.setup] Setup failed for argon40: Integration failed to initialize.
2021-10-05 17:02:50 ERROR (Thread-2) [root] Uncaught thread exception
Traceback (most recent call last):
File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
self.run()
File "/usr/local/lib/python3.9/threading.py", line 910, in run
self._target(*self._args, **self._kwargs)
File "/usr/local/lib/python3.9/logging/handlers.py", line 1518, in _monitor
self.handle(record)
File "/usr/local/lib/python3.9/logging/handlers.py", line 1499, in handle
handler.handle(record)
File "/usr/local/lib/python3.9/logging/__init__.py", line 952, in handle
self.emit(record)
File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 192, in emit
entry = LogEntry(record, stack, _figure_out_source(record, stack, self.hass))
File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 99, in __init__
self.message = deque([record.getMessage()], maxlen=5)
File "/usr/local/lib/python3.9/logging/__init__.py", line 367, in getMessage
msg = msg % self.args
TypeError: %d format: a number is required, not NoneType`

Misiu, 2021-10-05:
I've set up a new dev instance running 2021.9.7, I've enabled I2C and the integration is working fine.
Could someone please try to enable I2C one more time and report back?
I've used this method: https://community.home-assistant.io/t/add-on-hassos-i2c-configurator/264167 for me it was the easiest one.

Misiu, 2021-10-05:
Do a hard reset. Pull the plug out off, reset from server control panel won't do. Also check if the Addon is running. I had to start it after I did a hard reset.

C0pywriting, 2021-10-05:
@Misiu 
Thanks for the quick help and the solution to the problem. 
On my device it already works. Tomorrow I have access to a 2 tes device with which I can try it again. 

I did it with the same plugin.

@maobuz 
can you send me the full log of the plugin?

[image]

C0pywriting, 2021-10-05:
@Misiu @maobuz 

For me it was a reboot over the system and remove the power cable once after that it worked 

[image]

maobuz, 2021-10-05:
crw--w----    1 root     tty         4,  22 Feb  2  2021 tty22
crw--w----    1 root     tty         4,  23 Oct  6 00:28 tty23
crw--w----    1 root     tty         4,  24 Feb  2  2021 tty24
crw--w----    1 root     tty         4,  25 Oct  6 00:28 tty25
crw--w----    1 root     tty         4,  26 Oct  6 00:28 tty26
crw--w----    1 root     tty         4,  27 Feb  2  2021 tty27
crw--w----    1 root     tty         4,  28 Feb  2  2021 tty28
crw--w----    1 root     tty         4,  29 Oct  6 00:28 tty29
crw--w----    1 root     tty         4,   3 Feb  2  2021 tty3
crw--w----    1 root     tty         4,  30 Oct  6 00:28 tty30
crw--w----    1 root     tty         4,  31 Feb  2  2021 tty31
crw--w----    1 root     tty         4,  32 Oct  6 00:28 tty32
crw--w----    1 root     tty         4,  33 Feb  2  2021 tty33
crw--w----    1 root     tty         4,  34 Oct  6 00:28 tty34
crw--w----    1 root     tty         4,  35 Feb  2  2021 tty35
crw--w----    1 root     tty         4,  36 Oct  6 00:28 tty36
crw--w----    1 root     tty         4,  37 Feb  2  2021 tty37
crw--w----    1 root     tty         4,  38 Oct  6 00:28 tty38
crw--w----    1 root     tty         4,  39 Oct  6 00:28 tty39
crw--w----    1 root     tty         4,   4 Feb  2  2021 tty4
crw--w----    1 root     tty         4,  40 Oct  6 00:28 tty40
crw--w----    1 root     tty         4,  41 Feb  2  2021 tty41
crw--w----    1 root     tty         4,  42 Oct  6 00:28 tty42
crw--w----    1 root     tty         4,  43 Oct  6 00:28 tty43
crw--w----    1 root     tty         4,  44 Oct  6 00:28 tty44
crw--w----    1 root     tty         4,  45 Oct  6 00:28 tty45
crw--w----    1 root     tty         4,  46 Oct  6 00:28 tty46
crw--w----    1 root     tty         4,  47 Oct  6 00:28 tty47
crw--w----    1 root     tty         4,  48 Oct  6 00:28 tty48
crw--w----    1 root     tty         4,  49 Oct  6 00:28 tty49
crw--w----    1 root     tty         4,   5 Oct  6 00:28 tty5
crw--w----    1 root     tty         4,  50 Oct  6 00:28 tty50
crw--w----    1 root     tty         4,  51 Oct  6 00:28 tty51
crw--w----    1 root     tty         4,  52 Oct  6 00:28 tty52
crw--w----    1 root     tty         4,  53 Oct  6 00:28 tty53
crw--w----    1 root     tty         4,  54 Oct  6 00:28 tty54
crw--w----    1 root     tty         4,  55 Oct  6 00:28 tty55
crw--w----    1 root     tty         4,  56 Oct  6 00:28 tty56
crw--w----    1 root     tty         4,  57 Oct  6 00:28 tty57
crw--w----    1 root     tty         4,  58 Oct  6 00:28 tty58
crw--w----    1 root     tty         4,  59 Oct  6 00:28 tty59
crw--w----    1 root     tty         4,   6 Oct  6 00:28 tty6
crw--w----    1 root     tty         4,  60 Oct  6 00:28 tty60
crw--w----    1 root     tty         4,  61 Oct  6 00:28 tty61
crw--w----    1 root     tty         4,  62 Oct  6 00:28 tty62
crw--w----    1 root     tty         4,  63 Oct  6 00:28 tty63
crw--w----    1 root     tty         4,   7 Oct  6 00:28 tty7
crw--w----    1 root     tty         4,   8 Oct  6 00:28 tty8
crw--w----    1 root     tty         4,   9 Oct  6 00:28 tty9
crw-rw----    1 root     audio     166,   0 Oct  6 00:44 ttyACM0
crw-rw----    1 root     audio     204,  64 Oct  6 00:28 ttyAMA0
crw-------    1 root     root        5,   3 Oct  6 00:28 ttyprintk
crw-------    1 root     root       10, 239 Feb  2  2021 uhid
crw-------    1 root     root       10, 223 Feb  2  2021 uinput
crw-rw-rw-    1 root     root        1,   9 Feb  2  2021 urandom
drwxr-xr-x    3 root     root            60 Feb  2  2021 v4l
crw-------    1 root     root      247,   0 Oct  6 00:28 vc-mem
crw-------    1 root     root      242,   0 Feb  2  2021 vchiq
crw-------    1 root     root      246,   0 Feb  2  2021 vcio
crw-rw----    1 root     tty         7,   0 Oct  6 00:28 vcs
crw-rw----    1 root     tty         7,   1 Oct  6 00:28 vcs1
crw-rw----    1 root     tty         7,   2 Oct  6 00:28 vcs2
crw-rw----    1 root     tty         7,   3 Oct  6 00:28 vcs3
crw-rw----    1 root     tty         7,   4 Oct  6 00:28 vcs4
crw-rw----    1 root     tty         7,   5 Oct  6 00:28 vcs5
crw-rw----    1 root     tty         7,   6 Oct  6 00:28 vcs6
crw-rw----    1 root     tty         7, 128 Oct  6 00:28 vcsa
crw-rw----    1 root     tty         7, 129 Oct  6 00:28 vcsa1
crw-rw----    1 root     tty         7, 130 Oct  6 00:28 vcsa2
crw-rw----    1 root     tty         7, 131 Oct  6 00:28 vcsa3
[... 7 lines cut ...]
crw-rw----    1 root     tty         7,  67 Oct  6 00:28 vcsu3
crw-rw----    1 root     tty         7,  68 Oct  6 00:28 vcsu4
crw-rw----    1 root     tty         7,  69 Oct  6 00:28 vcsu5
crw-rw----    1 root     tty         7,  70 Oct  6 00:28 vcsu6
crw-------    1 root     root       10,  63 Feb  2  2021 vga_arbiter
crw-------    1 root     root       10, 137 Feb  2  2021 vhci
crw-------    1 root     root       10, 238 Feb  2  2021 vhost-net
crw-rw----    1 root     netdev     81,   0 Feb  2  2021 video10
crw-rw----    1 root     netdev     81,   2 Feb  2  2021 video11
crw-rw----    1 root     netdev     81,   6 Feb  2  2021 video12
crw-rw----    1 root     netdev     81,   1 Feb  2  2021 video13
crw-rw----    1 root     netdev     81,   3 Feb  2  2021 video14
crw-rw----    1 root     netdev     81,   4 Feb  2  2021 video15
crw-rw----    1 root     netdev     81,   5 Feb  2  2021 video16
crw-------    1 root     root       10, 130 Feb  2  2021 watchdog
crw-------    1 root     root      249,   0 Feb  2  2021 watchdog0
crw-rw-rw-    1 root     root        1,   5 Feb  2  2021 zero
brw-rw----    1 root     disk      254,   0 Feb  2  2021 zram0
brw-rw----    1 root     disk      254,   1 Feb  2  2021 zram1
brw-rw----    1 root     disk      254,   2 Feb  2  2021 zram2
/dev/i2c-1
/dev/i2c-0 /dev/i2c-1 /dev/i2c-10 /dev/i2c-11
Found i2c access!  Nothing to do!  You can remove this add-on.

ok after reboot this is a log  but nor see argon40

C0pywriting, 2021-10-05:
@maobuz 
please set the following settings and trigger the automation. 

then the fan should go on 

/config/configuration.yaml

argon40:

sensor:
  - platform: command_line
    name: CPU Temp
    command: "cat /sys/class/thermal/thermal_zone0/temp"
    unit_of_measurement: "°C"
    value_template: "{{ value | multiply(0.001) | round(1) }}"

/config/automations.yaml

- id: '1'
  alias: Set fan speed higher 55C to 40%
  description: ''
  trigger:
  - platform: numeric_state
    entity_id: sensor.cpu_temp
    above: '55'
    for:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  condition: []
  action:
  - service: argon40.set_fan_speed
    data:
      speed: 40
  mode: single
- id: '2'
  alias: Set fan speed under 50 C to 0%
  description: ''
  trigger:
  - platform: numeric_state
    entity_id: sensor.cpu_temp
    for:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
    below: '50'
  condition: []
  action:
  - service: argon40.set_fan_speed
    data:
      speed: 0
  mode: single
- id: '3'
  alias: Set fan speed higher 65C to 100%
  description: ''
  trigger:
  - platform: numeric_state
    entity_id: sensor.cpu_temp
    above: '65'
    for:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  condition: []
  action:
  - service: argon40.set_fan_speed
    data:
      speed: 100
  mode: single

maobuz, 2021-10-06:
> @maobuz please set the following settings and trigger the automation.
>
> then the fan should go on
>
> /config/configuration.yaml
>
> argon40:
>
> sensor:
>
> * platform: command_line
>   name: CPU Temp
>   command: [...]

apparently work but not sure ... 

i tested two automations but the fan not step apparentaly dow (sorry for my bad eng ...)

maobuz, 2021-10-06:
Work ! @Misiu  @C0pywriting  THX a lot!

maobuz, 2021-10-06:
[argon40_set_fan_speed.zip](https://github.com/Misiu/argon40/files/7293093/argon40_set_fan_speed.zip)
[image]

Misiu, 2021-10-14:
Thank you all for your help and patience.

closed 2021-10-14 by Misiu