Misiu/argon40 #30: No argon40 integration in core-2021.11.3

thierry-rhone, 2021-11-14:
Hi,
core-2021.11.3
supervisor-2021.10.8

I have copied argon40 folder from master branch
So I have the 4 files
custom_components/argon40/__init__.py
custom_components/argon40/const.py
custom_components/argon40/manifest.json
custom_components/argon40/services.yaml

I restarted home assistant twice, clear navigator cache but I don't have any argon integration in configuration add integration.

It is a compatibility problem

Misiu, 2021-11-14:
I think not. Did you add entry to configuration.yaml? It is needed because for now my component doesn't support config flow.

thierry-rhone, 2021-11-14:
Thank for your answer...
I added argon40: in my configuration.yaml, (and sensor and automation)
I have the message ::
The following integrations and platforms could not be set up:    argon40

Misiu, 2021-11-14:
You must enable I2C. This is the reason for that, message.

thierry-rhone, 2021-11-14:
It's ok with I2C
Thank's

Marie-Isere, 2021-11-15:
even if I no longer have the message "The following integrations and platforms could not be set up: argon40", when I call the set service in the development tools with data 40, I have this result

[image]

thierry-rhone, 2021-11-15:
even if I no longer have the message "The following integrations and platforms could not be set up: argon40", when I call the set service in the development tools with data 40, I have this result
(fan speed don't change)

[image]

Misiu, 2021-11-15:
@thierry-rhone try putting:
```
speed: 40
```
as service data.
This should do the trick.

closed 2021-11-16 by thierry-rhone