Misiu/argon40 #31: Following your directions but..... cant get the integration to show up.

ClintBritt, 2021-11-29:
I must be doing something stupid. Sorry new to the HA game.... but I have followed your directions exactly as I see them, but I can never see the Argon4 integration:

The directory structure is as follows:
[image]

But the integration list never has a Argon4 element:

[image]

What am I doing wrong?

Misiu, 2021-11-29:
Hi,

you must add `agron40:` to configuration.yaml (step 8 in readme) and enable I2C:
https://github.com/Misiu/argon40/issues/25#issuecomment-851439794
This should help with fixing the issue.
Please let me know if You have any problems.

squalorn, 2021-12-12:
Hi,
I am experiencing the issue. But I am completely new with HA, anything what is not said is ambigous for me. Then apologise for my silly question.
I followed all the steps but I do not understand add argon40: to my configuration.yaml.

Just write argon40: at the bottom of configuration as follows:
-----------
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

http:
  ssl_certificate: /ssl/fullchain.pem
  ssl_key: /ssl/privkey.pem
  
argon40: #??????
--------

and where do we need to write the following in section Config ?

Thx

closed 2022-01-20 by stale[bot]