Misiu/argon40 #32: Unable to load since 2022.6

Minims, 2022-06-06:
## Version of the custom_component

## Configuration

```yaml

2022-06-06 08:44:45 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration argon40 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-06-06 08:44:57 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration argon40: Unable to import component: cannot import name 'SMBus' from 'smbus' (/usr/local/lib/python3.9/site-packages/smbus/__init__.py)
2022-06-06 08:48:25 ERROR (MainThread) [homeassistant.components.automation.argon_one_fan_speed_0_50degc] Argon One: Fan speed 0% (<50°C): Error executing script. Service not found for call_service at pos 1: Unable to find service argon40.set_fan_speed
2022-06-06 08:48:25 ERROR (MainThread) [homeassistant.components.automation.argon_one_fan_speed_0_50degc] Error while executing automation automation.argon_one_fan_speed_0_50degc: Unable to find service argon40.set_fan_speed

```

## Describe the bug
The integration cannot be load because this python module cannot be found : 
```
Unable to import component: cannot import name 'SMBus' from 'smbus' (/usr/local/lib/python3.9/site-packages/smbus/__init__.py
```

## Debug log

```text

Add your logs here.

```

Minims, 2022-06-06:
I think this is related to :  https://github.com/home-assistant/architecture/blob/master/adr/0019-GPIO.md

Misiu, 2022-06-06:
@Minims yes, I think this is related, since from 2022.6 RPI.GPIO isn't loaded.
Please try adding https://github.com/thecode/ha-rpi_gpio, I hope this will help.
Report back and I'll add a proper description or will continue to investigate.

Minims, 2022-06-06:
It doesn't help, and I think this is normal.
This integration will be loaded in my custom_component folder, and I think your code will try to load the sambas module from HA core.

2 ideas to check : 

 - if ha-rpi_gpio make the job, maybe your code can load the lib from this repo ? 
 - you can add a fix version of `smbus`lib in your repo, update imports and  use it ?

viiy, 2022-07-07:
So, unable fix that?

Misiu, 2022-07-07:
Didn't have time (newborn at home), I'll try to find time as soon as I can

viiy, 2022-07-08:
> Didn't have time (newborn at home), I'll try to find time as soon as I can

Сongratulations!

viiy, 2022-07-08:
> GPIO need an update since 2022.07.0. See PR #33 If you have issue after upgrading HA, delete and remove the add-on from HACS. It will reinstall the libs. I think HA uninstall GPIO libs on upgrade

Unable to install package RPi.GPIO==0.7.1a4: error: subprocess-exited-with-error × python setup.py bdist_wheel did not run successfully. │ exit code: 1 ╰─> [15 lines of output] running bdist_wheel running build running build_py creating build creating build/lib.linux-aarch64-cpython-310 creating build/lib.linux-aarch64-cpython-310/RPi copying RPi/__init__.py -> build/lib.linux-aarch64-cpython-310/RPi creating build/lib.linux-aarch64-cpython-310/RPi/GPIO copying RPi/GPIO/__init__.py -> build/lib.linux-aarch64-cpython-310/RPi/GPIO running build_ext building 'RPi._GPIO' extension creating build/temp.linux-aarch64-cpython-310 creating build/temp.linux-aarch64-cpython-310/source gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fno-semantic-interposition -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/local/include/python3.10 -c source/c_gpio.c -o build/temp.linux-aarch64-cpython-310/source/c_gpio.o error: command 'gcc' failed: No such file or directory [end of output] note: This error originates from a subprocess, and is likely not a problem with pip. ERROR: Failed building wheel for RPi.GPIO error: subprocess-exited-with-error × Running setup.py install for RPi.GPIO did not run successfully. │ exit code: 1 ╰─> [17 lines of output] running install /usr/local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools. warnings.warn( running build running build_py creating build creating build/lib.linux-aarch64-cpython-310 creating build/lib.linux-aarch64-cpython-310/RPi copying RPi/__init__.py -> build/lib.linux-aarch64-cpython-310/RPi creating build/lib.linux-aarch64-cpython-310/RPi/GPIO copying RPi/GPIO/__init__.py -> build/lib.linux-aarch64-cpython-310/RPi/GPIO running build_ext building 'RPi._GPIO' extension creating build/temp.linux-aarch64-cpython-310 creating build/temp.linux-aarch64-cpython-310/source gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fno-semantic-interposition -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/local/include/python3.10 -c source/c_gpio.c -o build/temp.linux-aarch64-cpython-310/source/c_gpio.o error: command 'gcc' failed: No such file or directory [end of output] note: This error originates from a subprocess, and is likely not a problem with pip. error: legacy-install-failure × Encountered error while trying to install package. ╰─> RPi.GPIO note: This is an issue with the package mentioned above, not pip. hint: See above for output from the failure.

Minims, 2022-07-08:
RPi.GPIO==0.7.1a4 need to be set to 0.7.1 like in my PR. Waiting for merge

ChrisHaPunkt, 2022-07-08:
Confirmed working again with updated dependency

Misiu, 2022-07-08:
Thank you for solving this issue. I'll merge the PR and release new version

Misiu, 2022-07-10:
A new release is on the way

viiy, 2022-07-11:
HA  2022.7.3 all work!
Thank you!

Misiu, 2022-07-11:
@Minims I'm waiting for your feedback before closing this issue.

closed 2022-07-11 by Misiu