Misiu/argon40 #35: Just a query

OptimusGREEN, 2022-12-31:
This would be better suited under discussions but I don't see the option.

Is this only for HA installed on the argon?

My HA is installed on another server but would like to control the remote argons fan from HA, is this possible with this?

Also, is this compatible with the argon eon?

Thanks

Misiu, 2023-01-11:
Hi there,
I'm sorry for not getting back to you sooner. I don't have EON to test the code.
My integration is intended to work with Home Assistant running on PI3/4 in Argon cases (or with Argon fans) and not with remote instances.
The code for controlling the fan is straightforward, so I think it would be possible to create a simple daemon in Python and control it remotely, but I don't have that much experience in Python to do that.

OptimusGREEN, 2023-01-13:
Thanks for the reply. I'll close this.

closed 2023-01-13 by OptimusGREEN