Misiu/argon40 #56: Error Loading Argon40 - HA startup

upsidedown76, 2023-10-28:
I use HA 2023.10.3 core installation on a PI4.

I did all the necessary steps - i2c is activated - checked in terminal and as well downloaded the i2c configurated which told me that i2c is activated. here results from terminal:
# lsmod | grep i2c
i2c_dev                20480  2
i2c_mux_pinctrl        16384  0
i2c_mux                16384  1 i2c_mux_pinctrl
i2c_bcm2835            16384  1

## Version of the custom_component - newest

## Configuration

```yaml

Add your logs here.

This error originated from a custom integration.

Logger: custom_components.argon40
Source: custom_components/argon40/__init__.py:73
Integration: argon40 (documentation, issues)
First occurred: 21:54:03 (1 occurrences)
Last logged: 21:54:03

Bad logger message: Error %d, %s accessing 0x%02X: Check your I2C address ((None, None, 26))
Traceback (most recent call last):
  File "/config/custom_components/argon40/__init__.py", line 73, in async_setup
    bus.write_byte(I2C_ADDRESS, 26)
  File "/usr/local/lib/python3.11/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
OSError: 121

Misiu, 2023-10-30:
Hi @upsidedown76,
sorry to hear you have problems using my integration.
I'm not aware of any problems with the recent versions of Home Assistant.
Which method of activating I2C did you use?
Do you have other I2C devices connected to your PI?

upsidedown76, 2023-10-30:
Hi,

I think I did it finally with the i2c configurator. And I have a external
hard drive mounted on one of the usb - but not sure if this has anything to
do with i2c. Am still a rookie with raspberry pi.

Best, Oliver

Misiu, 2023-10-30:
@upsidedown76,

so if I understand correctly, the issue is fixed and you can use the integration? If yes, then please close the issue.

> Am still a rookie with raspberry pi.

Same here :)

upsidedown76, 2023-10-30:
nope, sorry not to being clear - I finally did the i2c with the i2c
configurator add on.

I still have the same error. Do you use the Argon Fan Hat or do you use
Argon One?

Best,

Oliver

Misiu, 2023-10-30:
Hi,
I used Argon One V2 case (https://argon40.com/products/argon-one-v2-case-for-raspberry-pi-4).
Sadly I don't have access to HAT or the new M2 case. Maybe the address of the FAN changed.
Can you try Raspbian and the official script?
Copy can be found here: https://github.com/okunze/Argon40-ArgonOne-Script

upsidedown76, 2023-10-30:
Hi Tomasz,

I tired the Argon One plugin and that unfortunately does not work either.
As your instructions said that it works with the Fan Hat I was very happy
about it. So guess its not working. How do I find out the address of the
fan? Am happy to play around with the script, but need some help ;).

Best,

Oliver

Misiu, 2023-10-30:
you can try to contact Argon40 for help.
If I remember correctly both the case and the fan has the same address.
My case had some instructions inside the box.

You can try i2c-tools:
https://learn.adafruit.com/scanning-i2c-addresses/raspberry-pi

upsidedown76, 2023-10-30:
Thanks a lot! I only have the fan, not the case.

Best, Oliver

upsidedown76, 2023-10-30:
And i found a device on 50. But no clue where to amend this in your script.
Tried in line 73, but no success.

Best, Oliver

Misiu, 2023-10-30:
Try https://github.com/Misiu/argon40/blob/master/custom_components/argon40/__init__.py#L29
You should change 0x1A to 0x50 or 0x32 (not sure if results from i2c are in dec or hex)

Misiu, 2023-10-30:
Could you please add a picture of the hat or post a link to the product (HAT)?

upsidedown76, 2023-10-30:
Hi Tomasz,

Attached a picture (front and back) as well as a screenshot of the error
log in HA. This one I did not post in Github.

Best, Oliver

Misiu, 2023-10-30:
Could you please reply using GitHub?
When replying via email no attachments are added.

upsidedown76, 2023-10-30:
images included now
[image]
[image]
[image]

upsidedown76, 2023-10-30:
and here what i get when I go into root in terminal and then - it seems that i2c should be ok.
cd /dev 
ls
[image]

Misiu, 2023-10-31:
I just contacted Argon40 and they confirmed that they are still using the I2C address and codes from https://github.com/Argon40Tech/Argon-ONE-i2c-Codes
When I started developing this integration, the first thing I did was run the official Raspberry Pi OS (https://www.raspberrypi.com/software/) and use the official script from Argon40 (instructions should be in the box). Then when I confirmed that the case (or HAT in your case) works I switched to HA.
Did you try Pi OS? This way you will be sure that the HAT is working.

upsidedown76, 2023-10-31:
thanks a lot. I did not try Pi OS yet. Will do and let you know if it
works.

Misiu, 2023-11-30:
@upsidedown76 any updates?

Misiu, 2023-11-30:
let us keep this open until your issue is solved.
Thanks for the update

closed 2024-04-08 by Misiu