Misiu/argon40 #67: HomeAssistantType was used from argon40, this is a deprecated alias which will be removed in HA Core 2025.5

techillin, 2024-05-02:
Logger: homeassistant.helpers.typing
Källa: helpers/deprecation.py:206
Inträffade först: 09:49:15 (1 händelser)
Senast loggade: 09:49:15

HomeAssistantType was used from argon40, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'argon40' custom integration

Misiu, 2024-05-02:
Thank you for reporting this.
I'll try to release an update as soon as possible, but I must setup a dev env on PI4 first.

welcoMattic, 2025-02-25:
It is still an issue.

welcoMattic, 2025-04-07:
HA actual version is 2025.4.1, so 2025.5 is very close, and argon40 module is still in error. 
Any plans to release at least a dedicated version for 2025.5 HA?
Not critical, but if you need help in any way, let us know ;)

welcoMattic, 2025-05-08:
Not stale, still accurate

Misiu, 2025-05-08:
I have this ready on the PI5 branch. Will try to port the changes ASAP. PR is welcome

Misiu, 2025-05-08:
@welcoMattic can you please check my changes on your system?
https://github.com/Misiu/argon40/pull/88/files
It's just a matter of changing 5 lines of code.

welcoMattic, 2025-05-08:
I've just updated the file, restart my HA, and here are the error logs:

```
Logger: homeassistant.setup
Source: setup.py:422
First occurred: 9:36:01 PM (1 occurrences)
Last logged: 9:36:01 PM

Error during setup of component argon40: Failed to add edge detection
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 422, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/argon40/__init__.py", line 71, in async_setup
    GPIO.add_event_detect(shutdown_pin, GPIO.BOTH, callback=event_detect_callback)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Failed to add edge detection
```

welcoMattic, 2025-05-09:
Thank you, now it works!

Misiu, 2025-05-09:
> Thank you, now it works!

Thank you for the testing. Now I can merge it and release new version.

Melectro1, 2025-06-20:
I've also edited the lines in the two files and I can also confirm, it works!

a-27m, 2025-07-10:
@Misiu this seems ready for release, right? If not, how
can we help?

No pressure, it just would be super nice to have now, when the latest HA release removed deprecated stuff and broke argon.

Misiu, 2025-07-14:
@a-27m I'll try to release a new version in the coming days.
Sadly, missing time is the main reason.
Could you please confirm that https://github.com/Misiu/argon40/pull/89 works for you? It's basically the same change, but also with a version bump

Misiu, 2025-07-23:
@a-27m thank you for checking and for kind words :)
I just released version 0.0.6.

The issue should be fixed.

closed 2025-07-23 by Misiu