Misiu/argon40 #69: RuntimeError: Failed to add edge detection after update to HomeAssistantOS 12.3

ChrisHaPunkt, 2024-05-08:
## Version of the custom_component
0.0.5

## Configuration

```yaml
...
argon40:
...

```

## Describe the bug
After updating to [HomeAssistant OS 12.3](https://github.com/home-assistant/operating-system/releases/tag/12.3) the integration fails to load.

## Debug log

```text

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 404, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/argon40/__init__.py", line 71, in async_setup
    GPIO.add_event_detect(shutdown_pin, GPIO.BOTH, callback=event_detect_callback)
RuntimeError: Failed to add edge detection

```

Misiu, 2024-05-08:
The edge detection was added to allow button double click support.
If you have the integration locally you can try to comment that line and check if it helps.
I don't have HA on PI4 to check the recent version. I'm waiting for a new PI to arrive.

Misiu, 2024-05-09:
@ChrisHaPunkt any updates?
You can always create an issue in HAOS repo - https://github.com/home-assistant/operating-system/issues as the problem occurs after the update to 12.3

Misiu, 2024-05-09:
https://github.com/home-assistant/operating-system/issues/3353
https://github.com/thecode/ha-rpi_gpio/issues/255
not only my integration has problems with HA OS 12.3

there is an alternative package for GPIO communication (https://github.com/waveform80/rpi-lgpio), but I think I'll remove the button support for now and wait for ha-rpi_gpio issue to be solved first (we can't have both old nad new libraries installed at the same time)

shawnheide, 2024-05-09:
@Misiu, I have an rpi 4 and commented the line you referenced and the integration loaded. Thanks also for the references to the underlying issues.

IvansZuks, 2024-06-26:
@Misiu same for me. Commented out line 71 and integration loaded. RPI 4

maobuz, 2024-07-02:
HELP line 71 commented but the integration not detected 
[image]
[image]
[image]

ChrisHaPunkt, 2025-02-02:
Thanks @Misiu , would love to see an upstream solution in place instead of the (currenty working) workaround with the commented line.

closed 2025-05-06 by stale[bot]