Misiu/argon40 #74: Argon40 services not loaded anymore since at least HA 2024.6

Tibronx, 2024-07-12:
Hi,

It was working for years and now suddenly not recognized anymore - HA 2024.6 & 2024.7 got same behavior.
Argon40 services (set fan speed) are not loaded anymore, not available via developer tools either.
I suspect a change within HA that impact custom_components/argon40/__init__.py file.

Below logs:
2024-07-12 15:28:23.762 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration argon40 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-07-12 15:28:27.155 WARNING (MainThread) [homeassistant.components.meteo_france] 1 hour rain forecast not available. Thusy - Rhône-Alpes (74) - FR is not in covered zone
2024-07-12 15:28:30.067 WARNING (ImportExecutor_0) [homeassistant.helpers.typing] HomeAssistantType was used from argon40, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'argon40' custom integration
2024-07-12 15:28:30.075 ERROR (MainThread) [homeassistant.setup] Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 404, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/argon40/__init__.py", line 71, in async_setup
    GPIO.add_event_detect(shutdown_pin, GPIO.BOTH, callback=event_detect_callback)
RuntimeError: Failed to add edge detection
2024-07-12 15:28:40.758 ERROR (MainThread) [homeassistant.components.automation.argon_one_fan_speed_0_50degc_init] Argon One: Fan speed 0% (<50°C) - INIT: Error executing script. Service not found for call_service at pos 1: Service argon40.set_fan_speed not found
2024-07-12 15:28:46.005 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/usr/local/lib/python3.12/site-packages/pysiaalarm/data/sia_codes.json', 'rb') in /usr/local/lib/python3.12/site-packages/pkg_resources/__init__.py, line 1758: return open(self._fn(self.module_path, resource_name), 'rb') inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/aio/server.py", line 54, in handle_line
    event = self.parse_and_check_event(data)
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/base_server.py", line 64, in parse_and_check_event
    event = SIAEvent.from_line(line, self.accounts)
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/event.py", line 161, in from_line
    return SIAEvent(
  File "<string>", line 28, in __init__
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/event.py", line 266, in __post_init__
    self.set_sia_code()
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/event.py", line 106, in set_sia_code
    self.sia_code = _load_sia_codes().get(self.code)  # pylint: disable=E1101
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/data/data.py", line 44, in _load_sia_codes
    data = _load_data(FILE_SIA_CODES)
  File "/usr/local/lib/python3.12/site-packages/pysiaalarm/data/data.py", line 38, in _load_data
    stream = pkg_resources.resource_stream(__name__, file)
  File "/usr/local/lib/python3.12/site-packages/pkg_resources/__init__.py", line 1248, in resource_stream
    return get_provider(package_or_requirement).get_resource_stream(
  File "/usr/local/lib/python3.12/site-packages/pkg_resources/__init__.py", line 1758, in get_resource_stream
    return open(self._fn(self.module_path, resource_name), 'rb')

2024-07-12 15:33:30.670 ERROR (MainThread) [homeassistant.components.automation.argon_one_fan_speed_10_50degc] Argon One: Fan speed 10% (>50°C): Error executing script. Service not found for call_service at pos 1: Service argon40.set_fan_speed not found
2024-07-12 15:37:00.677 ERROR (MainThread) [homeassistant.components.automation.argon_one_fan_speed_10_50degc] Argon One: Fan speed 10% (>50°C): Error executing script. Service not found for call_service at pos 1: Service argon40.set_fan_speed not found
2024-07-12 15:38:40.478 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.home_to_gva_p51 is taking over 10 seconds

Tibronx, 2024-07-12:
Other logs that might help:

Logger: homeassistant.setup
Source: setup.py:404
First occurred: 3:28:30 PM (1 occurrences)
Last logged: 3:28:30 PM

Error during setup of component argon40
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 404, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/argon40/__init__.py", line 71, in async_setup
    GPIO.add_event_detect(shutdown_pin, GPIO.BOTH, callback=event_detect_callback)
RuntimeError: Failed to add edge detection

Misiu, 2024-07-13:
Hi, sorry for the late reply.
Take a look at https://github.com/Misiu/argon40/issues/69

The recent version of HAOS removed/changed GPIO access. I'm working on a new version of the integration with changed library.

Tibronx, 2024-07-15:
@Misiu Thanks ! I missed that one. Workaround worked.

maobuz, 2024-11-02:
Any news @Misiu  ? 
[image]

Misiu, 2024-11-08:
@maobuz I've set up a new dev environment using Pi5 and Argon ONE V3.
The working code for Pi5 is on a new branch - https://github.com/Misiu/argon40/tree/pi5

Could you show more logs?

maobuz, 2024-11-10:
> @maobuzHo impostato un nuovo ambiente di sviluppo usando Pi5 e Argon ONE V3. Il codice funzionante per Pi5 è su un nuovo ramo - https://github.com/Misiu/argon40/tree/pi5
> 
> Potresti mostrare altri log?

Dettagli registro (AVVISO)
Registratore: homeassistant.helpers.typing
Fonte: helpers/deprecation.py:222
Prima occorrenza: 23:01:25 (1 occorrenze)
Ultima registrazione: 23:01:25

HomeAssistantType was used from argon40, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'argon40' custom integration

i have remove old installation and install a new versione now, but i have a same problem

Misiu, 2024-11-11:
@maobuz what hardware are you using? Pi 4 or Pi 5?
I have a working version for Pi 5 on a new branch - https://github.com/Misiu/argon40/tree/pi5
You need edit [manifest.json](https://github.com/Misiu/argon40/blob/pi5/custom_components/argon40/manifest.json) and [__init__.py](https://github.com/Misiu/argon40/blob/pi5/custom_components/argon40/__init__.py).
That is, if you are using Pi 5.

If you are using Pi 4 for now remove this line in your local copy: https://github.com/Misiu/argon40/blob/main/custom_components/argon40/__init__.py#L71

This line is responsible for button press logic.

maobuz, 2024-11-11:
> @maobuz what hardware are you using? Pi 4 or Pi 5? I have a working version for Pi 5 on a new branch - https://github.com/Misiu/argon40/tree/pi5 You need edit [...]

Hi Misiu :) 

Device Pi4  

i commented the line 71 but the integration not work. 

is impossible set fan (- service: argon40.set_fan_speed ) is not present

Misiu, 2024-11-11:
are you sure you have I2C Enabled?
Take a look at this comment: https://github.com/Misiu/argon40/issues/75#issuecomment-2464357465
I've installed `HassOS I2C Configurator`, disabled protection mode, started the addon, and restarted my Pi (I unplugged the power cable) twice. Only then I get the valid output.

maobuz, 2024-11-11:
> are you sure you have I2C Enabled? Take a look at this comment: [#75 (comment)](https://github.com/Misiu/argon40/issues/75#issuecomment-2464357465) I've installed `HassOS I2C Configurator`, disabled protection mode, started the addon, and restarted my Pi (I unplugged the power cable) twice. Only then I get the valid output.

'AddonManager.install' blocked from execution, supervisor needs to be updated first .... When install It

Misiu, 2024-11-11:
what HAOS version are you running? did you disable protection mode?

maobuz, 2025-01-05:
> HAOS version are you running? did you disable protection mode?

[image]

any news to resolve the problem ?

Misiu, 2025-01-05:
Not sure what else I can do here. I setup a fresh system on PI4 and PI5 and i had no problem running the integration.
How did you enable I2C?

maobuz, 2025-01-05:
> I2C
i follow this guide 
https://community.home-assistant.io/t/add-on-hassos-i2c-configurator/264167
[image]

closed 2025-04-25 by stale[bot]