Misiu/argon40 #92: Error %d, %s accessing 0x%02X: Check your I2C address ((None, None, 26))

Sagi-tab, 2025-08-04:
Hi Misiu!

Thanks for your hard work on this integration 🙏🏼
I started getting the above alerts yesterday after a restart I performed. No other changes I can thing of.

Some necessary info:
> Installation method Home Assistant OS
> Core 2025.7.4
> Supervisor 2025.07.3
> Operating System 16.0
> Frontend 20250702.3

Using Argon ONE M.2 for Raspberry Pi 4

I2C installation method - The official way (and I verified it using the add-on, just in case)
Integration installed through HACS

Seems like this is stemming from: Source: custom_components/argon40/__init__.py:73 -
`bus.write_byte(I2C_ADDRESS, 10)`

[image]

> ```
> This error originated from a custom integration.
> 
> Logger: custom_components.argon40
> Source: custom_components/argon40/__init__.py:73
> integration: argon40 (documentation, issues)
> First occurred: 15:36:12 (1 occurrence)
> Last logged: 15:36:12
> 
> Bad logger message: Error %d, %s accessing 0x%02X: Check your I2C address ((None, None, 26))
> ```

Please let me know if I can assist with collecting more info or if any further information is needed to debug this.
Thanks!

Misiu, 2025-08-04:
@Sagi-tab, if I understand correctly, the integration worked and stopped working.
Did you update the HAOS? I found an issue with HAOS 16.0 - https://github.com/home-assistant/operating-system/issues/4189 it is related to 1-Wire, but it is GPIO-based, so maybe a related issue.
Please confirm which version of HAOS you are using. Thank you.

Misiu, 2025-08-04:
I just noticed you are using HAOS 16. Can you roll back to 15.2? Just to confirm everything works.

Misiu, 2025-08-05:
Could you try to test a fresh HAOS 16 on a spare SD card?
I'm waiting for a new PI to try to recreate the issue on HAOS 15.2 and 16.

Sagi-tab, 2025-08-05:
Unfortunately I don't have any other available rbp to test it on.
Any chance this has something to do with the fact I'm running on rbp with
nvme instead of an SD?

nimnal, 2025-09-05:
Good day! Tell me, do I understand correctly that at the moment this automation does not work on the current version?
I have it installed HAOS:
Type: Home Assistant OS
Core: 2025.8.3
Supervisor: 2025.08.3
Operating System: 16.0
User interface: 20250811.1
...and it does not start and does not work in any form, although I2C successfully starts and works on the host.
I tried installing the integration manually, as well as adding it via HACS.
Best regards!